wpilibsuite/allwpilib #1129: New controller API

calcmogul, 2018-06-03:
PIDController's focus on SISO-only is making the addition of features like Kv/Ka feedfowards, composition, superposition of outputs, and 1 DOF, multi-state motion profiles much more difficult. The features definitely don't mesh as well as they should, and a lot of that is the result of PIDController's assumptions about the kind of system it is controlling. For example, it should be possible to specify position, velocity, or acceleration references through one call via a motion profile, but the current API has no way to pass that information through to a feedforward (see discussion in #978).

After some internal discussion, we determined it would be easier to design something from the ground up with MIMO controllers in mind that has specializations for the SISO controller case for convenience.

calcmogul, 2018-06-03:
List of requirements I drafted a while ago:

Requirements
-----
* Feature parity with PIDController
  * Can set and return reference and gains
  * Can return error
  * Can set error tolerance
  * Can return whether controller is on target
  * Integration with Sendable for setting gains
* Composable with other controllers
  * Done previously by inheriting from PIDWrite(). PIDController::PIDWrite() called SetSetpoint().
* Customizable feedforward algorithm
  * Could use Nu feedforward by default (u = Nu * r) which is analogous to Kv and Ka
  * How to support nonlinear feedforwards? Constant matrices don't work for that.
  * Perhaps use callback approach which can use reference or not
* Can configure filters for each output
  * MIMO state-space does this with state observer (it uses outputs to correct state estimate from model). Lack of a model makes this approach unusable.
  * LinearFilter, PIDSource, std::function<double()> overloads?
* Extendable to MIMO (e.g., drivetrain with differential drive with two control inputs and two outputs)
* Outputs of two controllers with the same control inputs can be superimposed
  * e.g., position and heading controllers of differential drive
* Asynchronous and synchronous versions
  * Main difference is whether values are passed in manually or pulled automatically by a Notifier with callbacks

PID controller feature analogs
-----
* Integral anti-windup
  * I-zone
    * MIMO: SetErrorRangeForIntegral()
      * Integral state is zeroed outside of this range.
  * MIMO: SetIntegralStateRange()
* SetInputRange()
  * MIMO: Output y range wrapping
* SetOutputRange()
  * MIMO: Control input u clipping
* SetPID()
  * MIMO: SetGains()
* SetSetpoint()
  * MIMO: SetReference()
    * Sets reference vector r
    * Overload for wpi::ArrayRef
* GetSetpoint()
  * MIMO: GetReference(int state)
* GetError()
  * MIMO: GetError(int state)
* SetPIDSourceType()/GetPIDSourceType()
  * MIMO: inherent to outputs of system and states specified by the user
  * N/A
* SetTolerance() et al.
  * MIMO: SetAbsoluteTolerance(int state, double tolerance)
  * MIMO: SetPercentTolerance(int state, double tolerance)
  * Note that deltaTolerance isn't needed because that's just the tolerance on the velocity state
* OnTarget()
  * MIMO: bool AtReference()
    * True means all states are on target (except integral state)
* Reset()
  * MIMO: Resets state and output vectors, then sends zero vector to control inputs

The general version can use a vector as the state that contains position and its derivatives. SISO PIDController wrapper around this would be possible, but things that feed into the controller like motion profiles would still need to use the vector version of SetReference().

PeterJohnson, 2019-07-30:
Fixed by #1300.

closed 2019-07-30 by PeterJohnson