wpilibsuite/allwpilib #1137: wpiHal shared library

pjreiniger, 2018-06-06:
TL;DR I can't compile and load my own native code that interacts links against the hal AND use wpilibj

Pre-2019 updates, when building the wpilibj native library, two+ versions were created; one for athena, which statically compiled the wpiutil class in, and one for windows (/native platform) that compiled it dynamically.  This allowed my simulator to load the wpiutil, wpihal, and wpilibJni (name?), and still be able to compile other native libraries that used the same wpihal library.  All hal symbols, nothing more or less, were contained in that DLL and I only loaded it once.

Now, as far as I can tell, since the JNI calls are done in the hal project, two libraries get published; the cpp version which I can dynamically link with, or the java version, which includes the JNI symbols but does not export everything needed to link other libraries against.

I don't know what the best thing to do is, maybe add a new artifact which is the JNI calls dynamically linked in addition to the static one, similar to before

ThadHouse, 2018-06-06:
The cpp artifact also exports the jni entry points. We need better documentation, but for all the native projects with jni, the -cpp artifact has all of it's dependencies shared linked, whereas the -jni has it's dependencies statically linked. The jni ones are meant to be embedded into jars for apps like shuffleboard to be self contained. Normally we do recommend the cpp artifacts.

ThadHouse, 2018-06-06:
Replaced with #1138. This is more of a documentation issue, and not an actual binary issue.

closed 2018-06-06 by ThadHouse