wpilibsuite/allwpilib #1160: Add Command.schedule() method

tervay, 2018-06-17:
In the 2018 docs, there is reference for a method that doesn't exist.

https://wpilib.screenstepslive.com/s/currentCS/m/java/l/599745-scheduling-commands

```
How to schedule a command from within a running command
Commands can be scheduled by calling the schedule() method (Java) or Schedule() method (C++) on a command instance. This will cause the command to be added to the currently running set of commands in the scheduler. This is often useful when a program needs to conditionally schedule one command or another. The newly scheduled command will be added to a list of new commands on this pass through the run method of the scheduler and actually will run the first time on the next pass through the run method. Newly created commands are never executed in the same call to the scheduler run method, always queued for the next call which usually occurs 20ms later.
```

However, there is no `schedule` method in the `Command` class. Unless I'm misinterpreting the docs, this method would be very useful to us.

auscompgeek, 2018-06-17:
I imagine these docs are meant to refer to Scheduler#add(Command).

AustinShalit, 2018-06-17:
Thank you for the bug report.  I have updated the documentation to correctly state that the method to call is the `start()` method.  You should use the this method to schedule commands.

closed 2018-06-17 by AustinShalit