wpilibsuite/allwpilib #1192: Support: Changing baud rate of CAN bus

gerth2, 2018-07-18:
Is there any interface or mechanism available for altering the baud rate of the CAN Bus? On our RIO with the 2018 latest release, we're seeing a 1Mbps rate. However, we're looking to experiment with a device that is (currently) hardcoded to operate at 250kbps. 

From digging through code, it seems most of the CAN communication funnels through [NI-provided API's](https://github.com/wpilibsuite/allwpilib/blob/master/ni-libraries/include/FRC_NetworkCommunication/CANSessionMux.h), but I'm not seeing something among them to set a baudrate.

Thanks!

ThadHouse, 2018-07-18:
No, there is no way to change the baud rate. All devices on the bus need to be at the same rate, and all FRC can devices run at 1MHz. So changing that value will break everything else in the can system.

gerth2, 2018-07-18:
Fair enough, it's a Non-FRC device we're working with. Thanks! We'll work the other end of the problem.

closed 2018-07-18 by gerth2