wpilibsuite/allwpilib #1221: Add PPA release for Ubuntu 18.04

Ewpratten, 2018-07-27:
There appears to be no way to get the FRC toolchain on 18.04 and I can not find a way to build it. Could someone add it to launchpad? Or if there is a way to build the toolchain for linux, I would love to know.

PeterJohnson, 2018-07-27:
@byteit101 will need to add it to launchpad.  If you want to build it yourself, it's a bit tricky, but the repo is here: https://github.com/wpilibsuite/toolchain-builder (there's a pending PR there to add Docker support that shows things more step-by-step).

Ewpratten, 2018-07-28:
ok.
I found that i can use the ubuntu old-releases repo and an old version of libmpfr to get it to work for now.

Ewpratten, 2018-07-31:
For your info, there will have to be some trickery to get it to work on 18.xx and up, because libmpfr no longer is available in the ubuntu repos.

Ewpratten, 2018-07-31:
I just grabbed a .deb from the ubuntu archives and used a tool called gdebi to install it.

PeterJohnson, 2018-08-07:
Packages should be up now if you want to try it.

Ewpratten, 2018-08-07:
looks good! Thanks!

closed 2018-08-08 by PeterJohnson