wpilibsuite/allwpilib #1233: Invalid JavaDoc

AustinShalit, 2018-07-28:
Right now we have some invalid JavaDoc comments:

```
/Users/austinshalit/git/allwpilib/wpilibj/src/main/java/edu/wpi/first/wpilibj/I2C.java:351: warning - @pre is an unknown tag.
/Users/austinshalit/git/allwpilib/wpilibj/src/main/java/edu/wpi/first/wpilibj/InterruptHandlerFunction.java:20: warning - Tag @link: reference not found: java.util.concurrent.atomic
/Users/austinshalit/git/allwpilib/wpilibj/src/main/java/edu/wpi/first/wpilibj/PWM.java:136: warning - @pre is an unknown tag.
/Users/austinshalit/git/allwpilib/wpilibj/src/main/java/edu/wpi/first/wpilibj/PWM.java:136: warning - @pre is an unknown tag.
/Users/austinshalit/git/allwpilib/wpilibj/src/main/java/edu/wpi/first/wpilibj/PWM.java:149: warning - @pre is an unknown tag.
/Users/austinshalit/git/allwpilib/wpilibj/src/main/java/edu/wpi/first/wpilibj/PWM.java:149: warning - @pre is an unknown tag.
/Users/austinshalit/git/allwpilib/wpilibj/src/main/java/edu/wpi/first/wpilibj/PWM.java:165: warning - @pre is an unknown tag.
/Users/austinshalit/git/allwpilib/wpilibj/src/main/java/edu/wpi/first/wpilibj/PWM.java:165: warning - @pre is an unknown tag.
/Users/austinshalit/git/allwpilib/wpilibj/src/main/java/edu/wpi/first/wpilibj/PWM.java:165: warning - @pre is an unknown tag.
/Users/austinshalit/git/allwpilib/wpilibj/src/main/java/edu/wpi/first/wpilibj/PWM.java:165: warning - @pre is an unknown tag.
/Users/austinshalit/git/allwpilib/wpilibj/src/main/java/edu/wpi/first/wpilibj/PWM.java:165: warning - @pre is an unknown tag.
/Users/austinshalit/git/allwpilib/wpilibj/src/main/java/edu/wpi/first/wpilibj/PWM.java:180: warning - @pre is an unknown tag.
/Users/austinshalit/git/allwpilib/wpilibj/src/main/java/edu/wpi/first/wpilibj/PWM.java:180: warning - @pre is an unknown tag.
/Users/austinshalit/git/allwpilib/wpilibj/src/main/java/edu/wpi/first/wpilibj/PWM.java:180: warning - @pre is an unknown tag.
/Users/austinshalit/git/allwpilib/wpilibj/src/main/java/edu/wpi/first/wpilibj/PWM.java:180: warning - @pre is an unknown tag.
/Users/austinshalit/git/allwpilib/wpilibj/src/main/java/edu/wpi/first/wpilibj/InterruptHandlerFunction.java:20: warning - Tag @link: reference not found: java.util.concurrent.atomic
/Users/austinshalit/git/allwpilib/wpilibj/src/main/java/edu/wpi/first/wpilibj/InterruptHandlerFunction.java:20: warning - Tag @link: reference not found: java.util.concurrent.atomic
```

sciencewhiz, 2018-07-28:
We had defined @pre as a custom javadoc tag. That probably got lost somewhere in the reorg. http://users.csc.calpoly.edu/~jdalbey/205/Resources/customtags

AustinShalit, 2018-07-29:
All of the `@pre` warnings were solved by #1238