wpilibsuite/allwpilib #1258: Feature Request: Make build meta-data available at runtime

gerth2, 2018-07-30:
In response to some [ongoing discussion on CD](https://www.chiefdelphi.com/forums/showthread.php?p=1777626#post1777626).

Provide gradle & runtime support for recording information about the software that was built, which can be retrieved at runtime (possibly automatically published to Network Tables).

Information which could be useful:
 - Build time/date
 - Build PC Name & User ID 
 - Git uncommitted files in working tree/Index (True/False) (if available)
 - Commit SHA1 (if available)
 - Git Branch (if available)
 - Git Tags (if any available)

Design: At build time, add gradle target to populate this info into a properties file. Properties file is deployed to the roboRIO with the main .jar. A class will have the ability to read this info and return strings based on the file contents, if available.

gerth2, 2018-07-31:
I have a [sample implementation](https://github.com/RobotCasserole1736/VsCodeTestProject/blob/master/build.gradle#L119) in our testing area for reference.

AustinShalit, 2020-09-13:
Closing in favor of: https://github.com/wpilibsuite/frc-docs/issues/783

closed 2020-09-13 by AustinShalit