wpilibsuite/allwpilib #1260: Unclear variable naming in MecanumDrive

jakebacker, 2018-07-31:
On line 119 of the MecanumDrive class, the parameter names go (`ySpeed, xSpeed, zRotation`) as seen below.
https://github.com/wpilibsuite/allwpilib/blob/0a0d9245e29b2210c636a263a307d321213b5985/wpilibj/src/main/java/edu/wpi/first/wpilibj/drive/MecanumDrive.java#L119
For most things, the order is `x, y, z`. This can (somehow) cause functionality issues in regards to inverted motors. On a project I am working on, I set the speeds as `y, x, z` as the variable names suggested. I also inverted motors for whatever reason and ended up with a robot that moved perfectly in every direction except turning. After 2 hours of debugging, I finally figured out that if all motors were un-inverted and the `x` and `y` drive parameters were swapped, the robot would perform correctly.

calcmogul, 2018-07-31:
That's due to how your electrical team wired the robot and the gearbox direction. These classes assume a positive voltage rotates the motor clockwise as viewed from the inside of the robot chassis.

calcmogul, 2018-07-31:
Also, the docs state that this class uses the NED axes convention, which means positive y is forward, positive x is right, and positive z is down. This means a positive z input results in clockwise rotation if the motors spin the directions the class expects.

closed 2018-09-17 by calcmogul