wpilibsuite/allwpilib #1263: Add option for constant feedforward term to PIDController for velocity control

Oblarg, 2018-08-08:
As covered in the [drive characterization whitepaper](https://www.chiefdelphi.com/media/papers/3402) we published on Chief Delphi last year, the "correct" form of the feedforward for most velocity control in FRC is not `Kf*setpoint`, but rather `Kf*setpoint + sgn(setpoint)*Kconst`, for some constant `Kconst`.  This correctly accounts for constant dry frictional losses in the mechanism, which otherwise can cause a sizeable nonlinearity at small setpoints.

Given that knowledge of this is becoming more widespread, I think it would be very beneficial to build this functionality into PIDController rather than requiring teams to implement it themselves, as it is not a particularly "heavy" change and can offer a significant improvement in performance.  Additionally, it would greatly facilitate the most common potential use-case of drive control, as [the current solution for PID drive with the WPILib drive objects](https://github.com/calcmogul/allwpilib/commit/7e1ea8c8a2e4fc3c8c8a5081c109136b760eed90) would make it quite difficult to change the feedforward implementation of the underlying PIDController.  Alternatively, it could be built into ClosedLoopMotor instead.

calcmogul, 2018-08-08:
I'm not a fan of ClosedLoopMotor's design either precisely because changing the controller it uses internally is basically impossible. I wrote it for API design exploration purposes. Adding new features to PIDController in general has been getting really difficult, and wrapping something like ClosedLoopMotor around it is proving to be painful. I'm working on a replacement for PIDController in https://github.com/wpilibsuite/allwpilib/projects/7.

For this change though, it would fit well with this PR that replaces Kf with Kv and Ka: https://github.com/wpilibsuite/allwpilib/pull/978. We closed it in anticipation of the controller rewrite because we didn't want to cause unnecessary churn in PIDController.

Oblarg, 2018-08-09:
Ah, that makes sense.  I suppose that the way to address all of these concerns will probably take clearer form as the replacement takes shape, then.

PeterJohnson, 2019-07-30:
Fixed by #1300.

closed 2019-07-30 by PeterJohnson