wpilibsuite/allwpilib #1277: ntcore RPC entries  spec non-compliant

Redrield, 2018-08-18:
In attempting to work on [nt-rs](https://gitlab.com/Redrield/nt-rs) to get RPC calls working correctly and integrated into the client, I built upon a logging ntcore server given to me by someone on the FRC discord. When attempting to run a pre-built client targeted at that server, the networking stack crashed and the application was stuck. After sniffing packets sent between two applications written with ntcore, this is what I found

```
10 -- Entry Assignment
07 -- LEB128 string length
72 70 63 54 65 73 74 -- "rpcTest" (entry name)
20 -- RPC definition entry type
00 00 -- Entry ID
00 01 -- Entry sequence number
01 00 -- ??? (Entry value)
03 -- next packet: Server Hello Complete
```

After more digging into ntcore, I found that when serializing an RPC definition entry value, ntcore will write a string into the socket and stop. (The string passed into CreateRpc).
This behaviour is absolutely contrary to what is written in the specification for [RPC definition data](https://github.com/wpilibsuite/allwpilib/blob/master/ntcore/doc/networktables3.adoc#remote-procedure-call-definition-data), and the malformed data is what was causing the crash.

I have created a workaround. However due to this issue nt-rs will not be able to function with RPC in a way that i'd like until ntcore is spec compliant.

PeterJohnson, 2018-08-19:
If you want to use RPCv1, ntcore splits the RPC definition logic into two parts: the NetworkTableEntry itself does not store the full complexity of the underlying RPC definition.  You need to use `nt::PackRpcDefinition()` to build the RPC definition from a `nt::RpcDefinition` structure and then provide that to `nt::CreateRpc()`.  Similarly, the `nt::CallRpc()` and `nt::GetRpcResult()` functions take raw values rather than the fully specified parameters, which have to use `nt::PackRpcValues()` (or `nt::UnpackRpcValues()` for RPCv1. This allows multiple RPC versions to be used by user code (e.g. RPCv1 is just one possibility).

The ntcore object-oriented C++ API defaults to RPCv0, which does need to be documented in the spec.  RPCv0 simplifies the RPC definition entry to be just a single 0 byte (e.g. no parameter or result specifications) and RPCv0 calls always send and return a single raw value (which falls back nicely to the `nt::CallRpc()` and `nt::GetRpcResult()` std::string input and output).

Redrield, 2018-08-19:
@PeterJohnson Put together some quick code that uses the APIs you describe, seems to be working, at least from what I can tell by the raw bytes of the RPC definition. Is there a way that I can immediately distinguish RPCv0 at the network level, so that I can read enough bytes to discard it?

PeterJohnson, 2018-08-19:
Sure, per the spec the first bytes of the definition is 1 for RPCv1.  For RPCv0 it’s 0.

Redrield, 2018-08-19:
If that's the case how is ntcore emitting that packet above? the first byte of the value is a 1, despite being RPCv0

Redrield, 2018-08-19:
RPCv0 is identical to RPCv1 with an empty string procedure name in the opening bytes. I'm not sure if that would be common but it's still possible.

PeterJohnson, 2018-08-19:
No, it's not.  RPCv1 definition starts with a byte value of 1.  So RPCv0 definition is just a single 00 byte, RPCv1 definition with an empty procedure name would be 01 01 00, followed by the rest of the definition.

Redrield, 2018-08-19:
From ntcore_cpp.cpp
```cpp
std::string PackRpcDefinition(const RpcDefinition& def) {
  WireEncoder enc(0x0300);
  enc.Write8(def.version);
  enc.WriteString(def.name);
...
```
A procedure name value of "" would yield 0 for the length followed by nothing. Thus resulting in 01 00 ...

PeterJohnson, 2018-08-19:
Whoops you're right.  But that's still different than just 00.

Redrield, 2018-08-19:
Indeed, but I have no guaranteed way of knowing whether the server sent an RPCv0 (because the first byte is 01 from LEB length encoding), or if it sent an RPCv1 with an empty procedure name.

It seems like the problem is that a version 3 server could theoretically send either. Perhaps a more ubiquitous API around RPCv1 and the deprecation of RPCv0 would be a solution to this.

PeterJohnson, 2018-08-19:
The encoding is the LEB128 length of the entire definition followed by the RPC definition.  So for RPCv0 the value is 01 (the LEB128 length of the definition, which is 1) 00 (version 0).  For RPCv1 it's going to be xx (the LEB128 length of the definition) 01 (version 1) xx (LEB128 length of the procedure name) etc.  You do NOT need to parse the RPC definition contents to parse the message, as the LEB128 length of the overall definition (regardless of RPC version) is provided.

PeterJohnson, 2018-08-19:
The whole reason for versioning the RPC separately was to allow for the possibility of different RPC versions to be available without changing the overall protocol revision.  If you don't understand a particular RPC version you can simply ignore those RPC entries.

PeterJohnson, 2018-08-20:
Documentation has been updated with info on the v0 format.

closed 2018-08-20 by PeterJohnson