wpilibsuite/allwpilib #1307: Support roboRIO 'USER' button

kasemir, 2018-09-11:
The roboRIO has a 'USER' button, right next to the 'RESET' button.
This can be useful to trigger for example a self-test right on the robot.
The USER button state can be read via `HALUtil.getFPGAButton()`, but that's a little hard to guess for a newcomer, and it's not a full `Button` with support for `Command`s.

Would it make sense to add this to the library?

```
/** Support for the 'USER' Button on the roboRIO.
 * 
 *  <p>
 *  Example Usage:
 *  
 *  <pre>
 *  Command command = new SomeCommand();
 *  Button user = new USERButton();
 *  // Start command when button is pressed,
 *  // stop command when button is released
 *  user.whenActive(command);
 *  </pre>
 */
public class USERButton extends Button
{
    /** @return <code>true</code> when the 'USER' button is pressed, otherwise <code>false</code> */
    public static boolean isPressed()
    {
        // It's a little hard to guess that this is the way to read the 'USER' button
        return HALUtil.getFPGAButton();
    }

    @Override
    public boolean get()
    {
        return isPressed();
    }
}
```

Or is this too specific to the roboRIO, since the next incarnation of the FRC hardware may not have a USER button?

auscompgeek, 2018-09-11:
You want [`RobotController.getUserButton()`](http://first.wpi.edu/FRC/roborio/release/docs/java/edu/wpi/first/wpilibj/RobotController.html#getUserButton--).

kasemir, 2018-09-11:
Ah, very good, that's better than `HALUtil.getFPGAButton()`. Still, it's just a boolean method, no `Button` that can trigger `Command`s. So would it make sense to create a full Button?

auscompgeek, 2018-09-11:
If you ask me, it probably wouldn't be a great idea.

- The potential usecases are very narrow. You wouldn't be able to use the USER button mid-match.
- Wrapping any boolean function in a Button is a few lines of code.
- Future hardware may or may not have a USER button.

Mind you, this is just the opinion of a random person on the internet.