wpilibsuite/allwpilib #1315: Incorrect DifferentialDrive documentation

tervay, 2018-09-18:
https://wpilib.screenstepslive.com/s/currentCS/m/java/l/914148-driving-a-robot-using-differential-drive#curvature-drive

>Like Arcade Drive, the Curvature Drive mode is used to control each side of the drivetrain independently (usually with an individual joystick axis controlling each). The difference is that the rotation control is attempting to control radius of curvature instead of rate of heading change.

From my understanding, Arcade Drive does not control each side of the drivetrain independently. Arcade controls throttle and a heading change, while Tank controls each side independently.

calcmogul, 2018-09-18:
"Forward movement and rotational degrees of freedom" is probably a better phrase than "side of the drivetrain".

Kevin-OConnor, 2018-09-18:
Yep, that's a very strange sentence as neither Arcade nor Curvature control each side independently

Kevin-OConnor, 2018-09-18:
Fixed to match the Arcade description above it

closed 2018-09-18 by Kevin-OConnor