wpilibsuite/allwpilib #1318: NidecBrushless doesn't inherit from PWMSpeedController despite being a PWM-commanded SpeedController

Matthew-Reynolds, 2018-09-20:
The `NidecBrushless` class is unique from most of the other speed controllers in that, in addition to a PWM signal to drive the speed & direction of the motor controller, it also has an enable line, as required by the Nidec Dynamo. As such, since it has this extra behaviour, instead of extending `PWMSpeedController`, it extends `SpeedController`.

This has a few impacts:
  - A lot of duplicated code. The implementation of `SpeedController` is identical except:
    - `getDescription()` prepends "Nidec" rather than "PWM"
    - `set(double)` needs to set both the DIO and PWM, rather than just the PWM
    - `stopMotor()` and `disable() need to set both the DIO and PWM to safe states, rather than just the PWM
    - `get()` varies slightly due to the composition rather than inheritance of `PWM` (below)
 - `NidecBrushless` interacts with `PWM` via composition, while all `PWMSpeedController`s use inheritance (though the additional `SafePWM`).
    - Makes the `getChannel()` function a simple pass-through to the internal PWM object.
 - Since `NidecBrushless` doesn't use `SafePWM` (Which implements `MotorSafety`), it implements `MotorSafety` itself. This leads to a lot of duplicated code:
   -  The implementation of `MotorSafety` is indentical except for:
      - `getDescription()` differs (See above)
      - `stopMotor()` differs (See above)

This isn't meant to be a comprehensive list, but I believe my point is clear. There is a LOT of duplicated code and redundancy in the `NidecBrushless`, and I think semantically it ought to be a `PWMSpeedController` as, at the end of the day, it is a speed controller being commanded with PWM.

Is there a reason for this structure that I'm missing? If not, I'm happy to open a PR changing the inheritance hierarchy of the class.

Matthew-Reynolds, 2018-09-20:
Perhaps is due to the safety issues discussed in [issue 1319](https://github.com/wpilibsuite/allwpilib/issues/1319), but looking over things quickly I don't see where that would be an issue

ThadHouse, 2018-09-20:
The Nicec is not a PWM speed controller like the other devices. It does not use standard PWM servo control signals for setting its duty cycle. So it uses the PWM servo signal, but only to send an enable disable signal. It really shouldn't just inherit PWMSpeedController, because it really just isn't one. All the control signals are completely different. So I don't think this is a change we would like, because its semantically invalid, even though it would reduce a little bit of code duplication.

closed 2018-09-20 by ThadHouse

Matthew-Reynolds, 2018-09-20:
I understand that the Nidec is different and therefore implemented differently, but I think I misunderstood the PWMSpeedController. I suppose the point of confusion here is that the PWMSpeedController is really more accurately for Servo-Style PWM Speed Controllers, while I was thinking of it as generically any speed controller commanded with a PWM signal. That seems to be why I classified the Nidec as semantically the same, while the WPILib classifies it as semantically distinct, regardless of implementation. Am I understanding correctly?

ThadHouse, 2018-09-20:
Yeah that's right. All of the other pwm speed controllers are serving signal controlled, which is why the base class was added a few years ago. The Nidec is an entirely different control scheme.

Matthew-Reynolds, 2018-09-20:
Makes sense, thanks for clarifying.

Perhaps a language change in the class's header comment would clarify that for others who get confused as I did, maybe describing the PWMSpeedController as a "Common base class for all Servo-Style PWM Speed Controllers" or something along those lines. Happy to make that change if it would be welcomed. Is there any particular/standard language that describes this style of PWM speed controller?

PeterJohnson, 2018-09-21:
I’ve heard the terms RC servo or hobby servo.  Wikipedia calls it simply “servo control” but that’s a little too short for my taste.  “Servo-style PWM Speed Controller” works for me.