wpilibsuite/allwpilib #1319: NidecBrushless uses DIO and PWM backwards

Matthew-Reynolds, 2018-09-20:
`NidecBrushless` contains a DIO and a PWM so that it can send both a direction+speed and an enable to the Dynamo. However, the PWM signal is generated on the DIO pin, and the enable line is driven by the PWM pin. This seems very backwards, so I'd like to change this, but changing it has the significant impact of changing the wiring needed to run the Dynamo. That being said, according to the usage statistics, only 8 teams used the Nidec Dynamo last year, so perhaps this isn't the end of the world.

Is there a reason this was implemented in this way? Is it safe for me to make this change, knowing that it will result in a change in the wiring?

ThadHouse, 2018-09-20:
This cannot be changed because of safety reasons. The pwm gets disabled when the robot gets disabled the dio does not.

ThadHouse, 2018-09-20:
The nidec also requires a different pwm Style then the main pwms can output

closed 2018-09-20 by ThadHouse

Kevin-OConnor, 2018-09-20:
The PWM signal generated is a "true" full duty cycle PWM and does not match the Servo PWM paradigm implemented on the PWM pins.

Matthew-Reynolds, 2018-09-20:
Makes sense, thanks for explaining