wpilibsuite/allwpilib #1320: NidecBrushless Brake/Coast

Matthew-Reynolds, 2018-09-20:
From my understanding of the GreenDrive manual, we should be able to achieve brake/coast control using the enable line (pin 8).

That is, when disabling the motor (pin 8 high), the motor will coast. When the motor is enabled (pin 8 low) and commanded 0 (50% duty), it will brake, or at least most BLDC controllers would. I can't find any information about 0 switching frequency on the GreenDrive (Whether it tries to be 'smart' and also shut off the FETs or something) and I don't have access to a Nidec Dynamo, so I was hoping someone with experience with the motor could confirm this behaviour.

PeterJohnson, 2018-09-28:
In terms of control of the GreenDrive, this functionality is already present.  If you call set(0.5) the motor controller is enabled with 50% duty cycle.  If you call stopMotor(), the motor controller is disabled.  I can't confirm whether this results in what would be traditionally called "coast" mode, but the software control is present to command the motor controller either way.

closed 2018-09-28 by PeterJohnson

Matthew-Reynolds, 2018-09-28:
I agree that teams are capable of controlling the motor in this way now, but I was more discussing bringing the controller more in line with the other SpeedControllers, or at least with controllers like CTRE's TalonSRX and VictorSPX. Perhaps this isn't desired, as the NidecBrushless is already quite different from all the other SpeedControllers both in control method and the fact that it's a bldc rather brushed, but I was imagining providing a `SetNeutralMode()`[(line 175](https://github.com/CrossTheRoadElec/Phoenix-api/blob/master/cpp/src/MotorControl/CAN/BaseMotorController.cpp#L175)) much like CTRE's motor controllers. Then, depending on the mode, the control when the robot is disabled and when a 0 is commanded would reflect this mode, either coast or brake. Just a software mimicking of the coast/brake select most of these controllers have with a jumper or eeprom/button.

Come to think of it, given the implementation with the enable PWM line always being disabled when the robot is disabled, I suppose it wouldn't be possible to have the motor be in 'brake' mode when disabled since the GreenDrive's smaller pullup than the Rio's pulldown causes it to be disabled. I still believe it would be a useful addition for when teams `set(0)`.

<sup> Note: per the GreenDrive manual, "A high level (+5 Vdc) on the Enable input causes the GreenDrive to stop controlling the motor, thus allowing the motor to coast". Assuming the 50% duty causes the BLDC to cog and hold position, I believe the GreenDrive should have very similar coast/brake behaviour to our traditional brushed motor controllers. <sup>