wpilibsuite/allwpilib #1375: Alpha: build error related to cookies

gerth2, 2018-10-17:
Building a brand new Timed robot template produces the following error:

`
PS C:\Users\FIRSTUser\Desktop\testproject\Test_vscode_project> ./gradlew.bat build

> Configure project :
Invalid cookie header: "Set-Cookie: has_recent_activity=1; path=/; expires=Wed, 17 Oct 2018 01:19:02 -0000". Invalid 'expires' attribute: WPS C:\Users\FIRSTUser\Desktop\testproject\Test_vscode_project> ./gradlew.bat build
> Configure project :
Invalid cookie header: "Set-Cookie: has_recent_activity=1; path=/; expires=Wed, 17 Oct 2018 01:41:46 -0000". Invalid 'expires' attribute: Wed, 17 Oct 2018 01:41:46 -0000
Invalid cookie header: "Set-Cookie: _octo=GH1.1.1577944445.1539736906; domain=.github.com; path=/; expires=Sat, 17 Oct 2020 00:41:46 -0000". Invalid 'expires' attribute: Sat, 17 Oct 2020 00:41:46 -0000
Invalid cookie header: "Set-Cookie: logged_in=no; domain=.github.com; path=/; expires=Sun, 17 Oct 2038 00:41:46 -0000; secure; HttpOnly". Invalid 'expires' attribute: Sun, 17 Oct 2038 00:41:46 -0000
NOTE: You are using an ALPHA version of GradleRIO, designed for the 2019 Season!
This release uses the 2018 Core Libraries, however all tooling (GradleRIO + IDE support) is incubating for 2019
If you encounter any issues and/or bugs, please report them to https://github.com/wpilibsuite/GradleRIO

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring root project 'Test_vscode_project'.
> Failed to notify project evaluation listener.
   > Connection reset
* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full
insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 2s
`

Not sure why we're getting a cookie error.... and am not super well versed in what gradle's doing under the hood that would require this. Any way to disable the check or work around it?

This is on VSCode 1.28.1, Java Extension pack 0.4.0

Any additional versions or background info I can provide?

ThadHouse, 2018-10-17:
Hmm. What jdk are you using?

gerth2, 2018-10-17:
Wait I'm bad at instructions. Need this submitted here, or moved to https://github.com/wpilibsuite/GradleRIO ?

As to JDK:
`
PS C:\Users\FIRSTUser\Desktop\testproject\Test_vscode_project> java -version
java version "1.8.0_144"
Java(TM) SE Runtime Environment (build 1.8.0_144-b01)
Java HotSpot(TM) 64-Bit Server VM (build 25.144-b01, mixed mode)
`

`%JAVA_HOME%` is not set...

ThadHouse, 2018-10-17:
Hmm ok. But yeah this issue is likely the best in gradlerio. 

I haven't actually seen that error before. I'll look around.

gerth2, 2018-10-17:
Well, maybe have an answer anyway. We're at our build facility now, behind the corporate firewall. I just VPN'ed to my home for unrelated reasons, happened to re-run the build, and the error went away. I'll close out.

closed 2018-10-17 by gerth2