wpilibsuite/allwpilib #1385: I2C Communication Failure Causes Fatal Behaviour

liamkinne, 2018-10-21:
tldr: if an i2c transmission fails, the code gets stuck and is unrecoverable until the program is restarted.

To reproduce:
```c++
void Robot::TeleopInit() {
  I2C device(frc::I2C::Port::kOnboard, 0x30);
  device.Write(0x50, 0x10); // no device to ACK
  std::cout << "Anyone Home?\n";
}
```

Essentially there is no fail-safe for when I2C device doesn't respond. This nailed my team during an important match because if the code gets locked up during autonomous (e.g. pixy cam becoming unplugged), the robot code has to be restarted before you can do anything in teleop. This problem is hard to catch because there isn't any error related to I2C displayed, we tracked it down due to the motor control watchdog not being fed.

I'm happy to work on implementing a timeout fail-safe in the HAL, but I'd peoples thoughts on what should happen if such an event occurs. In some Arduino libraries a failed write returns an error code, and a failed write returns some default value. What's the wpi way of dealing with this? is there a way for the HAL to raise an exception that can be caught later?

Starlight220, 2020-09-14:
Can someone test this on a RIO? Both Java and C++.
This can't be tested in simulation as there isn't a "device not connected" situation in sim, which is what we're trying to test.

AustinShalit, 2021-03-13:
I just tested this using your sample code and was not able to reproduce. Going to close this issue. Please reopen if you have additional information.

closed 2021-03-13 by AustinShalit