wpilibsuite/allwpilib #1392: Ultrasonic SetAutomaticMode Joins on each call

AustinShalit, 2018-10-29:
Ultrasonic SetAutomaticMode Joins on each call which means the user could be waiting more than 0.1 seconds for the call to return.  We can make this much faster by doing some thread safety instead of joining the thread.

calcmogul, 2020-08-26:
Are you suggesting the ultrasonic thread should be kept running at all times and made a no-op if automatic mode is disabled? The only other option is starting/stopping the thread, as we do now.

AustinShalit, 2020-08-26:
I don't remember exactly why this was an issue but I am suggesting that we don't wait for the thread to finish by removing the join call here: 
https://github.com/wpilibsuite/allwpilib/blob/ee6a814576013f6b543f3040b746a3e0f76f84df/wpilibj/src/main/java/edu/wpi/first/wpilibj/Ultrasonic.java#L272

closed 2020-09-13 by AustinShalit