wpilibsuite/allwpilib #1402: opencv-cpp-3.4.3-15-windowsx86-64 not found in repo

brhea, 2018-11-01:
I downloaded the zip and compiled using gradlew... ntcore and wpiutil compile ok but cscore fails with error: 

Download http://first.wpi.edu/FRC/roborio/maven/development/edu/wpi/first/thirdparty/frc2019/opencv/opencv-cpp/3.4.3-15/opencv-cpp-3.4.3-15-windowsx86-64.zip

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':cameraserver:linkCameraserverReleaseSharedLibrary'.
> Could not resolve all files for configuration ':eduwpifirstthirdpartyfrc2019opencvopencv-cppwindowsx86-64'.
   > Could not download opencv-cpp-windowsx86-64.zip (edu.wpi.first.thirdparty.frc2019.opencv:opencv-cpp:3.4.3-15)
      > Could not get resource 'http://first.wpi.edu/FRC/roborio/maven/development/edu/wpi/first/thirdparty/frc2019/opencv/opencv-cpp/3.4.3-15/opencv-cpp-3.4.3-15-windowsx86-64.zip'.
         > Read timed out

I searched on maven for windowsx86-64 file and could not locate it. Is the 64 bit version not published?

I was trying to do a complete build on windows before trying on raspberry pi. With the same gradlew command run on the PI... e.g. download to pi and the run gradlew build ?

AustinShalit, 2018-11-01:
Try to run the build again.  The zip does exist.

brhea, 2018-11-01:
I tried again and received the same error... I'm just compiling the cscore part now...
e.g.   gradlew :cscore:build

ThadHouse, 2018-11-01:
There are some bugs with Gradle where sometimes the initial downloads fail. Keep running the build, and eventually they will succeed.

The build does not work directly on the raspberry pi. It's won't be able to find the dependencies it needs. However, we do now have a cross compiler for the pi, and we actually compile everything for the pi now as well. We are working on docs for this.

If you need it now, you should use the cmake build for the pi. It's designed for coprocessors to work correctly, and doesn't have the dependency issues the Gradle build has.

brhea, 2018-11-01:
Thanks... I'll keep trying.

I don't need the files immediately.... However, I would like to get vision processing working with the latest code to ensure everything works together next year. 

I really just need the wpiutil.so, cscore.so, and ntcore.so.... and the .so file for opencv that is compatible with cscore.

ThadHouse, 2018-11-01:
Ah. If you're compiling from source you need Java 11.

brhea, 2018-11-01:
I don't think I'm quite ready to make the jump to Java 11... That would break much of what I was using last year (.... and my IT department at work has not migrated to it yet). 

I did a quick search on maven and found the raspbian so files I need:
libopencv_java343.so
libcscore.so
libntcore.so
libwpiutil.so

I appreciate you providing the raspbian binaries. I was just trying to learn to generate these myself and train our students on how to keep these up to date. This really simplifies things, now I'll just go download the binaries when bug fixes or updates occur. 

For now I'll continue with training our new students using last year's platform and update to Java 11 and the new binaries at kickoff.

ThadHouse, 2018-11-01:
Note that if you're using Java with those libraries, the Java builds we provide on maven are 11 as well. So you would need 11 on the Pi to run the Java version of those libraries. For C++ you should be fine.

Can I close this?

brhea, 2018-11-01:
Yes please close... and thanks again.

closed 2018-11-01 by ThadHouse