wpilibsuite/allwpilib #1430: Add multi-motor diff drive example

AustinShalit, 2018-11-14:
Create an example for this team: https://www.chiefdelphi.com/forums/showthread.php?threadid=178528

epicDISZ, 2020-03-23:
This seems like the wrong link? that CD post is about problems with Eclipse...

calcmogul, 2020-03-23:
The first post shows code for a multi-motor differential drive that is blatantly incorrect. He's saying we should provide an example showing how to do it correctly.

Starlight220, 2020-04-20:
PR, incoming!

ItayZiv, 2020-04-20:
I’m quite sure there are (possibly other) examples that include 2 motors per side (I don’t remember which ones exactly)

sciencewhiz, 2021-09-19:
The following examples all use MotorControllerGroup with DifferentialDrive:
armbot, armbotoffboard, frisbeebot, gearsbot, gyrodrivecommands, hatchbotinlined, hatchbottraditional, pacgoat, ramsetecommand, statespacedifferentialdrivesimulation

The following examples use MotorControllerGroup directly, without DifferentialDrive
differentialdrivebot, differentialdriveposeestimator, ramsetecontroller, simpledifferentialdrivetrainsimulation

The following examples use DifferentialDrive, without MotorControllerGroup
arcadedrive, arcadedrivexboxcontroller, gettingstarted, gyro, romireference, shuffleboard, tankdrive, tankdrivexboxcontroller, ultrasonic, ultrasonicpid

closed 2021-09-24 by calcmogul