wpilibsuite/allwpilib #1446: Unhandled NullPointerException while "stopping live window mode"

phurley, 2018-11-27:
This is seems to be caused by this snippet of code:
```
    ShuffleboardTab tab = Shuffleboard.getTab("System Tests");
```
Where a student creates an empty tab. 

```
Error at frc.robot.Main.main(Main.java:27): Unhandled exception: java.lang.NullPointerException
	at edu.wpi.first.wpilibj.smartdashboard.SendableBuilderImpl.stopListeners(SendableBuilderImpl.java:120)
	at edu.wpi.first.wpilibj.smartdashboard.SendableBuilderImpl.stopLiveWindowMode(SendableBuilderImpl.java:139)
	at edu.wpi.first.wpilibj.livewindow.LiveWindow.setEnabled(LiveWindow.java:76)
	at edu.wpi.first.wpilibj.IterativeRobotBase.loopFunc(IterativeRobotBase.java:203)
	at edu.wpi.first.wpilibj.TimedRobot.startCompetition(TimedRobot.java:81)
	at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:261)
	at frc.robot.Main.main(Main.java:27)
```

SamCarlberg, 2018-11-27:
Please share your code.

phurley, 2018-11-27:
Sorry it was in a pretty large "base code for our team" project, which is public, but a mess for you to try and use. So I made a new template based on timed robot and after some testing found it is actually the NavX control causing the problem. In particular adding the line:

```
public static AHRS navx = new AHRS(SPI.Port.kMXP);
```
to robot.java (or as a member variable in a subsystem which is where we had it) will replicate the problem. I assume this is actually a bug in the Kauai lab code, so feel free to close it here. I will send them a message as well.

For reference sake here is the project:

[simpletest.tar.gz](https://github.com/wpilibsuite/allwpilib/files/2621998/simpletest.tar.gz)

PR #1463 (2018-12-08): Fix LiveWindow attempting to start listeners on uninitialized sendables
Additionally adds a defensive check in SendableBuilderImpl to avoid the NPE

Fixes #1446
closed 2018-12-08 by PeterJohnson