wpilibsuite/allwpilib #1449: GearsBot building error

Ramtech59Sensei, 2018-11-28:
The method close() of type Claw must override or implement a supertype method [67109498]

The class Subsystem don't have a method Close() so compiler give error.
removing @Override will fix the problem

// original 
  /**
   * Set the claw motor to move in the close direction.
   */
  @Override               
  public void close() {
    m_motor.set(1);
  }

// fix
  /**
   * Set the claw motor to move in the close direction.
   */
  public void close() {
    m_motor.set(1);
  }

ThadHouse, 2018-11-28:
What environment are you seeing this in. Eclipse, the VS Code Alpha, or the VS Code Beta?

Ramtech59Sensei, 2018-11-28:
environment: VS Code Alpha (should give the same error in other env)
GearsBot example 
I check the new version repository and still there 
it is a simple error in the example file
gearsbot/subsystems/Wrist.java
but if not fix it will confuse students

ThadHouse, 2018-11-28:
In the alpha, this was broken for a different reason. The function is override able in the repo, and in the beta release (the function does the wrong thing, but that is a different issue.

closed 2018-11-28 by ThadHouse