wpilibsuite/allwpilib #1450: cscore build error

chuckb, 2018-11-28:
I am attempting to build from master and I get the following: 
```
> Task :cscore:compileCscoreBaseReleaseStaticLibraryCscoreBaseReleaseStaticLibraryCscoreWindowsCpp
COMCreators.cpp
Testing at beginning of ksmedia.h
C:\Program Files (x86)\Windows Kits\8.1\Include\um\ddraw.h(713): error C2011: '_DDPIXELFORMAT': 'struct' type redefinition
C:\Program Files (x86)\Windows Kits\8.1\Include\shared\ksmedia.h(6362): note: see declaration of '_DDPIXELFORMAT'
C:\Program Files (x86)\Windows Kits\8.1\Include\um\ddraw.h(2259): error C2079: '_DDSURFACEDESC::ddpfPixelFormat' uses undefined struct '_DDPIXELFORMAT'
C:\Program Files (x86)\Windows Kits\8.1\Include\um\ddraw.h(2302): error C2079: '_DDSURFACEDESC2::ddpfPixelFormat' uses undefined struct '_DDPIXELFORMAT'
C:\Program Files (x86)\Windows Kits\8.1\Include\um\strmif.h(13490): error C2011: 'tagTIMECODE_SAMPLE': 'struct' type redefinition
C:\Program Files (x86)\Windows Kits\8.1\Include\shared\ksmedia.h(5881): note: see declaration of 'tagTIMECODE_SAMPLE'
Testing at beginning of ksmedia.h
C:\Users\chuck\Documents\allwpilib\cscore\src\main\native\windows\COMCreators.cpp(48): warning C4838: conversion from 'DWORD' to 'int' requires a narrowing conversion

UsbCameraImpl.cpp
Testing at beginning of ksmedia.h
C:\Program Files (x86)\Windows Kits\8.1\Include\um\ddraw.h(713): error C2011: '_DDPIXELFORMAT': 'struct' type redefinition
C:\Program Files (x86)\Windows Kits\8.1\Include\shared\ksmedia.h(6362): note: see declaration of '_DDPIXELFORMAT'
C:\Program Files (x86)\Windows Kits\8.1\Include\um\ddraw.h(2259): error C2079: '_DDSURFACEDESC::ddpfPixelFormat' uses undefined struct '_DDPIXELFORMAT'
C:\Program Files (x86)\Windows Kits\8.1\Include\um\ddraw.h(2302): error C2079: '_DDSURFACEDESC2::ddpfPixelFormat' uses undefined struct '_DDPIXELFORMAT'
C:\Program Files (x86)\Windows Kits\8.1\Include\um\strmif.h(13490): error C2011: 'tagTIMECODE_SAMPLE': 'struct' type redefinition
C:\Program Files (x86)\Windows Kits\8.1\Include\shared\ksmedia.h(5881): note: see declaration of 'tagTIMECODE_SAMPLE'
Testing at beginning of ksmedia.h
```

Seems like an out of order include problem, but I doubt the build is broken. Is there some version of Windows platform SDK or directshow I need vs. what was installed by default with VS2015 Community?

ThadHouse, 2018-11-28:
We build with 2017 15.9, and the new cscore building was not tested with any visual studio versions before that. Also, we build with the windows 10 kits as well, so I don't know if it builds with the 8.1 kits.

chuckb, 2018-11-28:
Ah, sorry I thought the docs said 2015. I think I have the 10 kit sitting beside the 8...is there an easy directive I can try to see if I can build with 2015 before going back to 2017?

ThadHouse, 2018-11-28:
You can try messing with the include order in that file and see if you can get it building. If so pr the fix. 

We will update the readme.

chuckb, 2018-11-28:
I can go ahead and make doc change and include with PR. Also, Java 11 needed too, which I'll add. I had 8 and it did not work.

chuckb, 2018-11-28:
And just FYI...the beta-2 binaries must have been built with 2015, which is why I have it (due to my other unit testing work.) I had 2017 installed and of course 2015 would not install without uninstall of 2017. Arg.

ThadHouse, 2018-11-28:
Hmm that's odd, it should have been. They're the same version. Did your 17 install have c++?

chuckb, 2018-11-28:
Well that's a good question...maybe not. But same version, I don't think so...dependency walker shows wpihaljni.dll depending on MSVCP14D.DLL, which until I went back to 2015, was not on my machine.

ThadHouse, 2018-11-28:
Microsoft kept that version the same for 2017, and they're doing the same for 2019 as well. They wanted to keep binary compatibility.

And the debug libs would only be installed if the c++ workload was installed, not just visual studio.

chuckb, 2018-11-28:
Ok, C++ was prob not installed. So I do not create more work for everyone, I'll just go back to 2017 and try again.

chuckb, 2018-11-28:
It builds fine now...except that VSCode reports import resolution failures all over for import edu.wpi.first.hal.FRCNetComm, which seems strange.

ThadHouse, 2018-11-28:
In allwpilib? That happens because we generate those files, and sometimes the intellisense engine doesn't pick up those builds.

chuckb, 2018-11-28:
Yes, everywhere. So I guess there would be a setting needed to point to the generated files?

ThadHouse, 2018-11-28:
It's not really a setting. The only solution I have currently is to clear the java cache, and then close and repopen vs code. To do that, run the open java language server command, and in there you'll see a folder.  You want to delete up to the redhat.java folder to clear all the caches.