wpilibsuite/allwpilib #1481: Add constants to define widget types.

KeithRieck, 2018-12-09:
**Is your feature request related to a problem? Please describe.**  
The builder interface for Shuffleboard widgets allows the widget type to be specified in the "withWidget" method, but the widget type is specified as a String.  Users might not know all the potential widget types, and they might misspell them.

**Describe the solution you'd like**  
Please add text constants for the different widget types.  For instance
`public static final VOLTAGE_VIEW = "Voltage View";`

**Describe alternatives you've considered**  
An enumeration might also work, although I suspect we must accommodate new custom widget types.

SamCarlberg, 2018-12-09:
This is being addressed with #1468

SamCarlberg, 2019-08-01:
Fixed by #1468

closed 2019-08-01 by SamCarlberg