wpilibsuite/allwpilib #1502: LiveWindow PDP CAN Timeout

phurley, 2018-12-24:
Beta 4, newly imaged RIO, installed Phoenix library 5.11.1.0:

Without this line:

    private PowerDistributionPanel pdp = new PowerDistributionPanel(10);

There are no errors in the log, add that line and the following error repeats:

    Error at frc.robot.Main.main(Main.java:27): HAL: CAN Receive has Timed Out
    edu.wpi.first.hal.PDPJNI.getPDPChannelCurrent(Native Method)
    edu.wpi.first.wpilibj.PowerDistributionPanel.getCurrent(PowerDistributionPanel.java:66)
    edu.wpi.first.wpilibj.PowerDistributionPanel.lambda$initSendable$0(PowerDistributionPanel.java:119)
    edu.wpi.first.wpilibj.smartdashboard.SendableBuilderImpl.lambda$addDoubleProperty$3(SendableBuilderImpl.java:242)
    edu.wpi.first.wpilibj.smartdashboard.SendableBuilderImpl.updateTable(SendableBuilderImpl.java:95)
    edu.wpi.first.wpilibj.livewindow.LiveWindow.updateValues(LiveWindow.java:290)
    edu.wpi.first.wpilibj.IterativeRobotBase.loopFunc(IterativeRobotBase.java:262)
    edu.wpi.first.wpilibj.TimedRobot.startCompetition(TimedRobot.java:81)
    edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:261)
    frc.robot.Main.main(Main.java:27)`

Confirmed using Phoenix Tuner (PDP on Channel 10):

    Channel  0 (A):     0.00
    Channel  1 (A):     0.00
    Channel  2 (A):     0.00
    Channel  3 (A):     0.00
    Channel  4 (A):     0.00
    Channel  5 (A):     0.00
    Channel  6 (A):     0.00
    Channel  7 (A):     0.00
    Channel  8 (A):     0.00
    Channel  9 (A):     0.00
    Channel 10 (A):     0.00
    Channel 11 (A):     0.00
    Channel 12 (A):     0.00
    Channel 13 (A):     0.00
    Channel 14 (A):     0.00
    Channel 15 (A):     0.00
    Battery(V) :   12.89
       Temp(C) :   19.90
    FaultHardwareStatus : 0
    Current FAULTS : 0000000000000000
    FaultTemp : 0
    FaultVbat : 0
    Current FAULTS : 0000000000000000    (sticky)    
    StickyFaultTemp : 0
    StickyFaultVbat : 0
    
    "Light Device LED" to clear sticky faults.

    Build:Dec 16 2018 22:56:03
    Press "Refresh" to close.

ozrien, 2018-12-24:
I claim this is fixed in https://github.com/wpilibsuite/allwpilib/pull/1496.

Workaround is to disable live window for PDP.

Unrelated but... did you really change the device ID to '10'?  Its not a problem or anything, and its good for testing.  Just curious if there was a reason.   Sometimes I think about what multiple PDPs on a robot would achieve (more channels, more batteries, etc.)

phurley, 2018-12-25:
We always put the PDP at channel 10, our drive motors start at 1. No idea why, PCM is at 11 too -- who knows.

PeterJohnson, 2018-12-29:
Fixed by #1497.

closed 2018-12-29 by PeterJohnson