wpilibsuite/allwpilib #1528: Adding 3rd Party Libraries in VS Code

yellowjaguar5, 2019-01-04:
Hello!
I just installed all of the newly released WPIib things (and followed the instructions on [this page](https://wpilib.screenstepslive.com/s/currentCS/m/getting_started/l/999999-installing-c-and-java-development-tools-for-frc)). I'm now trying to add the NavX and CTRE 3rd party libraries to the robot project. I have not found any URLs or installers from Kawai Labs or CTRE to use in the `Manage Vendor Libraries` --> `Install New Libraries (online)` or `Install New Libraries (offline)` Please let me know if I'm doing something wrong or where I can find the installers/URLs/ways to get those libraries.

PeterJohnson, 2019-01-04:
They have not been posted yet.  Most likely today for CTRE.

Kevin-OConnor, 2019-01-04:
Hi there! I don't believe that either CTRE or Kauai have posted their 2019 season release yet. I know for at least CTRE they are doing some final testing and are on track to do so soon. I expect that Kauai is probably similar.

closed 2019-01-04 by yellowjaguar5

team2053tigertronics, 2019-01-04:
Will we be able to update directly from gradle or maven or have to redownload?

Kevin-OConnor, 2019-01-04:
Re-download what? The 3rd party libraries are not part of the WPILib distribution. They are installed as described on this page: https://wpilib.screenstepslive.com/s/currentCS/m/getting_started/l/682619-3rd-party-libraries

Note: There is also an option to have VSCode install the library from an online URL that is not documented on that page yet, but should be fairly straightforward (if the vendor supports it) and we aim to have documented soon.