wpilibsuite/allwpilib #1529: FRC VS Code 2019 shortcut isn't generated

cjdenio, 2019-01-05:
After installing WPILib 2019, the VS Code shortcut doesn't appear. The Shuffleboard and SmartDashboard shortcuts work fine, though.

ThadHouse, 2019-01-05:
Did you select a vscode to install, and then select the check box to actually install it? It won't generate the icon otherwise. You can also check to make sure vscode itself installed correctly by going to c:/users/public/frc2019 and making sure there is a vscode folder there.

cjdenio, 2019-01-05:
I'm really sorry, I didn't check the box (I thought it could use my existing VS Code installation). How might I uninstall it?

closed 2019-01-05 by cjdenio

ThadHouse, 2019-01-05:
You can just reinstall over the existing one. It will work just fine.

cjdenio, 2019-01-05:
Thank you! Just for future reference, how can this be uninstalled? (Can I just remove the `frc2019` folder?)

ThadHouse, 2019-01-05:
Yup. That and delete the shortcuts and you'll be good.

cjdenio, 2019-01-05:
Thank you! Sorry for opening this.