wpilibsuite/allwpilib #1532: Driverstation unable to open Shuffleboard and Smartdashboard

NamelessSuperCoder, 2019-01-06:
I have installed the latest 2019 suite, meaning the 2019 driver station, VSCode, the WPILIB extension associated with it, and the NI update all the while following the screen step live steps; however, when I open the driver station it gives me the error that it: "can not find script file "C:\WINDOWS\system32\%PUBLIC%\frc2019\tools\SmartDashboard.vbs"". The same error occurs when I try opening Shuffleboard but with Shuffleboard.vbs instead. If instead, I use the desktop shortcut to open it says: "can not find script file "C:\Users\mariu\Desktop\%PUBLIC%\frc2019\tools\SmartDashboard.vbs""
The actual files are located at "C:\Users\Public\frc2019\tools\"

ThadHouse, 2019-01-06:
We know, it's up to NI to fix that on their side.

NamelessSuperCoder, 2019-01-06:
Is there a way for me to send an issue to NI about the Driverstation?

ThadHouse, 2019-01-06:
They know about it. 

You can fix it in the driver station ini file, I would need to look that up though.

NamelessSuperCoder, 2019-01-06:
I have looked at my DriverStation.ini and the but It does not contain any property which deals with the location of Shuffleboard or SmartDashboard. This is what mine says:

[Driver Station]
server.app.propertiesEnabled=True
server.tcp.serviceName="My Computer/VI Server"
server.vi.propertiesEnabled=True
WebServer.TcpAccess="c+*"
WebServer.ViAccess="+*"
DebugServerEnabled=False
DebugServerWaitOnLaunch=False
SocketSetReuseAddr=True

sciencewhiz, 2019-01-21:
This is fixed in the NI FRC Update Suite 2019.1.0. http://www.ni.com/download/first-robotics-software-2017/7904/en/

closed 2019-01-21 by sciencewhiz