wpilibsuite/allwpilib #1535: Trouble building on Mac

kinahawi, 2019-01-08:
I’m encountering errors building WPILib (allwpilib-2019.1.1) an a Mac. See below.

I followed instructions in “Getting Started With the Control System” document to install and update the tools.
I have protobuf 3.6.1 and Gazebo 9.
Mac OSx 10.14.2

All help appreciated!

1: Task failed with an exception.

What went wrong:
Execution failed for task ‘:simulation:gz_msgs:linkGz_msgsReleaseSharedLibrary’.

> A build operation failed.

Linker failed while linking libgz_msgs.dylib.

See the complete log at: file:///Users/williamkinahan/frc2019/allwpilib-2019.1.1/simulation/gz_msgs/build/tmp/linkGz_msgsReleaseSharedLibrary/output.txt

Try:
Run with –stacktrace option to get the stack trace. Run with –info or –debug option to get more log output. Run with –scan to get full insights.

==============================================================================

2: Task failed with an exception.

What went wrong:
Execution failed for task ‘:simulation:gz_msgs:linkGz_msgsDebugSharedLibrary’.

> A build operation failed.

Linker failed while linking libgz_msgs.dylib.

See the complete log at: file:///Users/williamkinahan/frc2019/allwpilib-2019.1.1/simulation/gz_msgs/build/tmp/linkGz_msgsDebugSharedLibrary/output.txt

Try:
Run with –stacktrace option to get the stack trace. Run with –info or –debug option to get more log output. Run with –scan to get full insights.

==============================================================================

For the first failure, some more detailed info:

Task :simulation:gz_msgs:linkGz_msgsReleaseSharedLibrary FAILED
Undefined symbols for architecture x86_64:
“google::protobuf::DescriptorPool::InternalAddGeneratedFile(void const*, int)”, referenced from:
protobuf_float64_2eproto::AddDescriptorsImpl() in float64.pb.o
protobuf_bool_2eproto::AddDescriptorsImpl() in bool.pb.o
protobuf_frc_5fjoystick_2eproto::AddDescriptorsImpl() in frc_joystick.pb.o
protobuf_driver_2dstation_2eproto::AddDescriptorsImpl() in driver-station.pb.o
“google::protobuf::MessageFactory::InternalRegisterGeneratedFile(char const*, void (*)(std::__1::basic_string<char, std::__1::char_traits, std::__1::allocator > const&))”, referenced from:
protobuf_float64_2eproto::AddDescriptorsImpl() in float64.pb.o
protobuf_bool_2eproto::AddDescriptorsImpl() in bool.pb.o
protobuf_frc_5fjoystick_2eproto::AddDescriptorsImpl() in frc_joystick.pb.o
protobuf_driver_2dstation_2eproto::AddDescriptorsImpl() in driver-station.pb.o
“google::protobuf::UnknownFieldSet::ClearFallback()”, referenced from:
gazebo::msgs::Float64::Float64() in float64.pb.o
gazebo::msgs::Float64::Float64(gazebo::msgs::Float64 const&) in float64.pb.o
gazebo::msgs::Float64::~Float64() in float64.pb.o
gazebo::msgs::Float64::~Float64() in float64.pb.o
gazebo::msgs::Float64::~Float64() in float64.pb.o
gazebo::msgs::Float64::Clear() in float64.pb.o
gazebo::msgs::Float64::CopyFrom(google::protobuf::Message const&) in float64.pb.o
…

PeterJohnson, 2019-01-08:
We only test compile gazebo on linux, not mac.  Are you trying to get gazebo to work?  If not, deinstall protobuf and gazebo and the build should skip those bits.

kinahawi, 2019-01-08:
I am trying to get Gazebo to work to we can sim our solution ahead of time.

Sorry for the dumb question, but is Mac supported? There were installation instructions for Mac so I assumed it was.

ThadHouse, 2019-01-08:
We have never tested or attempted to build with the gazebo support on Mac. So the primary libraries are supported, but the gazebo stuff is not for any platform but Linux.

AustinShalit, 2019-08-17:
This is an unsupported configuration at the moment.

closed 2019-08-17 by AustinShalit