wpilibsuite/allwpilib #1541: ADXRS450_Gyro not functional

Bluethmann, 2019-01-11:
After updating roboRIO to 2019 image/firmware, the ADXRS450_Gyro class appears to be non-functional in c++.    

The example code (below) was tested on both 2019 and 2018 releases, with a TimedRobot template, adding a Gyro, then reading and displaying every 100 times through TeleopPeriodic.  It was tested using a FRC Gyro on SPI port CS0.  In both 2018 and 2019 releases, the gyro is recognized in the constructor and isn't recognized when the wrong port is specified.  The gyro works as intended with 2018 tools/rio and does not with 2019 tools/rio.

Note to test with 2018 image/firmware, the project will need to define _2018

Simple Robot.cpp
`/*----------------------------------------------------------------------------*/
/* Copyright (c) 2017-2018 FIRST. All Rights Reserved.                        */
/* Open Source Software - may be modified and shared by FRC teams. The code   */
/* must be accompanied by the FIRST BSD license file in the root directory of */
/* the project.                                                               */
/*----------------------------------------------------------------------------*/

#include "Robot.h"

#include <iostream>

#include <SmartDashboard/SmartDashboard.h>

void Robot::RobotInit() {
	m_chooser.AddDefault(kAutoNameDefault, kAutoNameDefault);
	m_chooser.AddObject(kAutoNameCustom, kAutoNameCustom);
	frc::SmartDashboard::PutData("Auto Modes", &m_chooser);

	m_gyro = new ADXRS450_Gyro(SPI::kOnboardCS0);

	m_count = 0;
	if(m_gyro != nullptr)
	{
		m_gyro->Calibrate();
	}

}

/**
 * This autonomous (along with the chooser code above) shows how to select
 * between different autonomous modes using the dashboard. The sendable chooser
 * code works with the Java SmartDashboard. If you prefer the LabVIEW Dashboard,
 * remove all of the chooser code and uncomment the GetString line to get the
 * auto name from the text box below the Gyro.
 *
 * You can add additional auto modes by adding additional comparisons to the
 * if-else structure below with additional strings. If using the SendableChooser
 * make sure to add them to the chooser code above as well.
 */
void Robot::AutonomousInit() {
	m_autoSelected = m_chooser.GetSelected();
	// m_autoSelected = SmartDashboard::GetString("Auto Selector",
	//		 kAutoNameDefault);
	std::cout << "Auto selected: " << m_autoSelected << std::endl;

	if (m_autoSelected == kAutoNameCustom) {
		// Custom Auto goes here
	} else {
		// Default Auto goes here
	}
}

void Robot::AutonomousPeriodic() {
	if (m_autoSelected == kAutoNameCustom) {
		// Custom Auto goes here
	} else {
		// Default Auto goes here
	}
}

void Robot::TeleopInit() {}

void Robot::TeleopPeriodic()
{
	if( (++m_count%100) == 0)
	{
		printf("gyro %.2f\n", m_gyro->GetAngle());
	}
}

void Robot::TestPeriodic() {}

START_ROBOT_CLASS(Robot)
`
with associate header Robot.h

`/*----------------------------------------------------------------------------*/
/* Copyright (c) 2017-2018 FIRST. All Rights Reserved.                        */
/* Open Source Software - may be modified and shared by FRC teams. The code   */
/* must be accompanied by the FIRST BSD license file in the root directory of */
/* the project.                                                               */
/*----------------------------------------------------------------------------*/

#pragma once

#include <string>

#include <SmartDashboard/SendableChooser.h>
#include <TimedRobot.h>
#ifdef _2018
#include <ADXRS450_Gyro.h>
#else
#include <frc/ADXRS450_Gyro.h>
using namespace frc;
#endif

class Robot : public frc::TimedRobot {
public:
	void RobotInit() override;
	void AutonomousInit() override;
	void AutonomousPeriodic() override;
	void TeleopInit() override;
	void TeleopPeriodic() override;
	void TestPeriodic() override;

private:
	frc::SendableChooser<std::string> m_chooser;
	const std::string kAutoNameDefault = "Default";
	const std::string kAutoNameCustom = "My Auto";
	std::string m_autoSelected;

	ADXRS450_Gyro *m_gyro;
	int m_count;

};
`

PeterJohnson, 2019-01-11:
This is a known issue with the 2019_v12 image.  See http://wpilib.screenstepslive.com/s/currentCS/m/getting_started/l/1028964-known-issues for a workaround.

Bluethmann, 2019-01-12:
Great thanks.  I must have missed this when I look through the issues after
kickoff.

Bill

PeterJohnson, 2019-01-19:
This is fixed in the 2019.1.0 NI update.

closed 2019-01-19 by PeterJohnson