wpilibsuite/allwpilib #1547: 'gradlew' is not recognized as an internal or external command

nathanrsxtn, 2019-01-12:
After running the WPILIB installer for current user only, building a java project returns the following error:
```
> Executing task: gradlew build  --offline -Dorg.gradle.java.home="C:\Users\Public\frc2019\jdk" <

'gradlew' is not recognized as an internal or external command,
operable program or batch file.
The terminal process terminated with exit code: 1

ThadHouse, 2019-01-12:
You likely have the wrong folder open in vscode. You need the folder with build.gradle and gradlew.bat open as the root folder. It can't be in a subfolder. this means if you have the file tree completely closed, you should see build.gradle. if you have to open a subfolder to see it, you have the wrong folder open and need to open the subfolder

nathanrsxtn, 2019-01-12:
I didn't have those files so I made a new sample project and built it, that fixed the problem.

closed 2019-01-12 by nathanrsxtn