wpilibsuite/allwpilib #1553: Java implementation does not use Realtime Specification  for Java RTSJ 1.0.2

vairamsvsjdo, 2019-01-16:
Java implementation does not use Realtime Specification  for Java (RTSJ) 1.0.2:
Specifications (
* https://jcp.org/aboutJava/communityprocess/first/jsr001/rtj.pdf
* https://www.aicas.com/rtsj/Version_1_0_2/
)
roboRIO OS is NI Linux Real-Time http://www.ni.com/white-paper/14627/en/, a realtime OS.

However, it appears that wpilib does not use RTSJ specifications. For example, VisionThread uses Thread instead of RealtimeThread or NoHeapRealtimeThread 
https://www.aicas.com/rtsj/Version_1_0_2/javax/realtime/RealtimeThread.html

https://www.aicas.com/rtsj/Version_1_0_2/javax/realtime/NoHeapRealtimeThread.html

https://github.com/wpilibsuite/allwpilib/blob/master/wpilibj/src/main/java/edu/wpi/first/wpilibj/vision/VisionThread.java

Kevin-OConnor, 2019-01-16:
This is not a change that will be made for 2019. We can evaluate pros/cons and make a determination on pursuing this next season

PeterJohnson, 2019-01-19:
There is no freely available JVM that supports RTSJ, so we won't be able to add support.

closed 2019-01-19 by PeterJohnson