wpilibsuite/allwpilib #1556: Task 'deploy' not found in root project

nathanrsxtn, 2019-01-18:
Building with Java returns successful, but deploying shows this error:
`FAILURE: Build failed with an exception.

* What went wrong:
Task 'deploy' not found in root project '2019-Hit-or-Miss'.

* Try:
Run gradlew tasks to get a list of available tasks. Run with --stacktrace option to get the stack trace. Run with --info or --debug option
to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 2s
The terminal process terminated with exit code: 1`

ThadHouse, 2019-01-18:
Are you trying to do a multi project setup? Can you post your build.gradle?

nathanrsxtn, 2019-01-18:
Putting everything in the root folder causes an error with building, which normally succeeds:
```
FAILURE: Build failed with an exception.

* Where:
Build file 'C:\Users\Scout-3\Documents\GitHub\2019-Hit-or-Miss\build.gradle' line: 53

* What went wrong:
A problem occurred evaluating root project '2019-Hit-or-Miss'.
> Could not find method pathfinder() for arguments [] on object of type org.gradle.api.internal.artifacts.dsl.dependencies.DefaultDependencyHandler.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full
insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 8s
The terminal process terminated with exit code: 1
```

ThadHouse, 2019-01-18:
Ah, I didn't notice, that's still an alpha project. Compile Pathfinder doesn't work anymore, and there are a ton of other changes from the alpha. Please create a new project and copy your code over.

nathanrsxtn, 2019-01-18:
Okay, that helped a lot. I'm getting some code errors now but I think it will deploy. Thanks!

closed 2019-01-18 by nathanrsxtn

kirupakarankumar, 2020-06-18:
Getting the following error while trying build helmpackage using gradle.
 
gradle helmPackage
Starting a Gradle Daemon (subsequent builds will be faster)

FAILURE: Build failed with an exception.

* What went wrong:
Task 'helmPackage' not found in root project 'pipeline-cdm'.

* Try:
Run gradle tasks to get a list of available tasks. Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 4s

This is the script :
stage('helmBuild') {
             steps {
        		 sh '''
        		 #TODO - HELM BUILD
                 echo "Skipping for now"
        		 '''
             }
        }
        stage('helmPackage') {
             steps {
        		 sh '''
        		 gradle helmPackage
        		 '''
             }
        }
stage('publishHelm') {
                    steps {
                    rtUpload (
                         serverId: "*******************",
                         spec:
                         """{
                            "files": [{
                             "pattern": "build/**********************
                             "target": "**********"
                             }]}"""
                         )

Any idea on this?

auscompgeek, 2020-06-18:
@kirupakarankumar What is helmpackage, and what does it have to do with FRC?

kirupakarankumar, 2020-06-19:
@auscompgeek Helm is a package management tool for kubernetes. I am configuring helm as package and build it  using gradle in my jenkinsfile and push it to private repo. I am getting the above error while triggering pipeline.

kirupakarankumar, 2020-06-19:
Also i guess we need some helm file in our repo which gradle will check while jenkins pipeline triggers.
I am not sure about the helm file.

auscompgeek, 2020-06-19:
@kirupakarankumar Well, Kubernetes is not useful at all in FRC, so you're definitely asking in the wrong place.