wpilibsuite/allwpilib #1575: Pathweaver crashing

Adhyyan1252, 2019-01-29:
This is the code I have implemented to follow the path using path weaver. Taken from https://wpilib.screenstepslive.com/s/currentCS/m/84338/l/1021631-integrating-path-following-into-a-robot-program

When I run the PathFollowInit() function, it crashes on the first command . Trajectory left_trajectory = PathfinderFRC.getTrajectory(k_path_name + ".right"); 
It dumps the core and outputs a SIGSEGV error on the roborio log. The path has only two waypoints and is only going forward so the path is definitely valid. 

private static final int k_ticks_per_rev = 360;
  private static final double k_wheel_diameter = 6 / 12.0;
  private static final double k_max_velocity = 4;
  private EncoderFollower m_left_follower;
  private EncoderFollower m_right_follower;
  private static final String k_path_name = "TestPath1";
  private Notifier m_follower_notifier;

  private void PathFollowerInit(){
    System.out.println("opening a file");
    Trajectory left_trajectory = PathfinderFRC.getTrajectory(k_path_name + ".right");
    System.out.println("opened one of them");
    Trajectory right_trajectory = PathfinderFRC.getTrajectory(k_path_name + ".left");
    System.out.println("opened");
    m_left_follower = new EncoderFollower(left_trajectory);
    m_right_follower = new EncoderFollower(right_trajectory);
    System.out.println("made followers");
    m_left_follower.configureEncoder(enc_l.get(), k_ticks_per_rev, k_wheel_diameter);
    // You must tune the PID values on the following line!
    m_left_follower.configurePIDVA(1.0, 0.0, 0.0, 1.000 / k_max_velocity, 0);

    m_right_follower.configureEncoder(enc_r.get(), k_ticks_per_rev, k_wheel_diameter);
    // You must tune the PID values on the following line!
    m_right_follower.configurePIDVA(1.0, 0.0, 0.0, 1 / k_max_velocity, 0);
    System.out.println("Configured the stuff");
    m_follower_notifier = new Notifier(this::followPath);
    m_follower_notifier.startPeriodic(left_trajectory.get(0).dt);
  }

  public void followPath() {
    if (m_left_follower.isFinished() || m_right_follower.isFinished()) {
      m_follower_notifier.stop();
    } else {
      double left_speed = m_left_follower.calculate(enc_l.get());
      double right_speed = m_right_follower.calculate(enc_r.get());
      double heading =  navx.getAngle();
      double desired_heading = Pathfinder.r2d(m_left_follower.getHeading());
      double heading_difference = Pathfinder.boundHalfDegrees(desired_heading - heading);
      double turn =  0.8 * (-1.0/80.0) * heading_difference;
      move(left_speed + turn, right_speed-turn);
    }
  }

PeterJohnson, 2019-01-29:
If the file doesn't exist, the getTrajectory call segfaults (yes, I consider this a bug).  Is the file where it can be loaded from?

Adhyyan1252, 2019-01-29:
I built the path from the Path Weaver app directly. It is in the directory ProjectName/PathWeaver/Paths/TestPath1

For reference this is the contents of the file
```
X,Y,Tangent X,Tangent Y,Fixed Theta,Name
3.891455680504592,22.713497502438937,10.0,0.0,true,Waypoint1
19.398671775530662,22.8693489204794,6.389908139658985,0.001,false,Waypoint2
```
Also in the ProjectName/PathWeaver/output file all the necessary pf1.csv files are there

PeterJohnson, 2019-01-29:
No, they can be at the top level.  You can use the Filesystem.getDeployDirectory() function to get the path of this on the robot side.

Adhyyan1252, 2019-01-29:
Thanks a lot for your help. I will deploy this on the robot today and check

closed 2019-02-26 by Kevin-OConnor