wpilibsuite/allwpilib #1581: Multiple PDP instances are allowed but break

ThadHouse, 2019-02-02:
Multiple are allowed to be created, however they override each other and break.

In actuality creating multiple should throw, but thats a more breaking change, so instead we should have singletons for this year and disallow multiple for next year.

Starlight220, 2020-03-16:
> Multiple are allowed to be created, however they override each other and break.
multiple created with different CAN IDs? Either way, I think there should be a decision about how to implement this (singleton or failing public constructor), and stick with it for this and next years. 
Personally, I believe that a failing public constructor is the way to go, as it is not a service (similar to CommandScheduler, DriverStation, etc.) but a device and should be consistent with other device classes.

Starlight220, 2020-04-19:
@calcmogul , anyone else, any ideas? If this is stale, should be closed.

ThadHouse, 2021-03-12:
This will be handled for 2022, whichever way we decide.

closed 2021-12-31 by PeterJohnson