wpilibsuite/allwpilib #1587: Add method to add CameraServer by name to Shuffleboard

auscompgeek, 2019-02-06:
With coprocessors for vision becoming popular, teams might find it beneficial to be able to add camera streams to Shuffleboard by name from their robot code (so all their layout code is in a single place).

SamCarlberg, 2019-02-06:
You mean beyond the method [ShuffleboardContainer.add(VideoSource)](http://first.wpi.edu/FRC/roborio/release/docs/java/edu/wpi/first/wpilibj/shuffleboard/ShuffleboardContainer.html#add(edu.wpi.cscore.VideoSource))?

eg:
```java
HttpCamera httpCamera = new HttpCamera("CoprocessorCamera", "frcvision.local:1181/stream.mjpg");
CameraServer.getInstance().addCamera(httpCamera);
Shuffleboard.getTab("Tab")
    .add(httpCamera);
```

auscompgeek, 2019-02-07:
Yeah. I... suppose that's one way of doing it. That would mean you'd be hardcoding the stream URL in the robot code though, unless you read the URLs off NT (but then you'd have to wait until the coproc has connected...).

Daltz333, 2019-03-11:
The code you posted @SamCarlberg does not work. It just shows a black screen. I confirmed the specified IP exists on the Raspberry Pi (static IPed) and I'm able to add the stream manually onto Shuffleboard.

auscompgeek, 2019-03-16:
Looks like the snippet above is missing the URL scheme (`http://`).

Daltz333, 2019-03-16:
Yeah. I forgot to mention that I figured that out after investigating networktables sources. There is also input URL inconsistencies between AxisCamera and HTTPCamera with one not requiring http and the other not.

auscompgeek, 2019-03-18:
> There is also input URL inconsistencies between AxisCamera and HTTPCamera with one not requiring http and the other does.

I'm pretty sure this is intentional. AxisCamera takes a hostname and will generate the full URL required for the stream for you (just prepending `http://` is obviously not enough as that will go to a webpage).

PeterJohnson, 2019-03-18:
Yes, AxisCamera just taking a hostname is intentional, to make it easy for users.  It's a fairly lightweight wrapper around HttpCamera.

PR #3850 (2022-01-01): [wpilib] Allow SendableCameraWrappers to take arbitrary URLs
Useful for adding cameras that are streamed from a coprocessor

Updated version of #1794.

Closes wpilibsuite/shuffleboard#582
Closes wpilibsuite/shuffleboard#598
Closes #1587
closed 2022-01-01 by PeterJohnson