wpilibsuite/allwpilib #1596: PIDBase setPIDSourceType and getPIDSourceType are package-private.

Ro5bert, 2019-02-11:
`setPIDSourceType(PIDSourceType)` and `getPIDSourceType()` in `edu.wpi.first.wpilibj.PIDBase` are package-private. Is this intentional? It appears that these are meant to be a way to change the input type for PID loop calculations (displacement vs. rate). Also, my IDE cannot find any usage of them in WPI lib.

auscompgeek, 2019-02-11:
Those methods are part of the PIDSource interface (i.e. using a PIDBase as a source for another PIDBase/PIDController).

If you want to change the input type of a sensor (such as an Encoder) to your PIDBase, then you should use those methods on that source object, not the PIDBase.

Ro5bert, 2019-02-11:
`PIDBase` does not implement `PIDSource`, though, and, if it did, those methods would have to be public (cannot weaken access privileges). Also, as I said above, my IDE (IntelliJ) cannot find usage of them in `wpilibj`.

auscompgeek, 2019-02-11:
Oh, right, a PIDBase can be used as a PIDOutput. What was I even talking about earlier.

The C++ PIDBase has those methods as public, so it seems like this was an oversight when PIDController was originally written.

PR #1599 (2019-02-14): Java PIDBase: Make setPIDSourceType and getPIDSourceType public
Fixes #1596.
commit 74f7ba04 (2019-02-14): Java PIDBase: Make setPIDSourceType and getPIDSourceType public (#1599)

Fixes #1596.
closed 2019-02-14 by PeterJohnson