wpilibsuite/allwpilib #1597: Infinite Recursion causes Weird Behavior

rselwyn, 2019-02-12:
Today when we were testing our software, we spent several hours debugging an issue seemingly without symptoms -- when we enabled our robot, the robot would *lose robot code and lose robot communication* (the robot would reconnect within about 3 seconds).  

All of this occurred without any stack trace.  The only thing that we saw was a "watchdog not fed within .02 seconds" warning.  Eventually, we found the root cause of the issue, but is there a better way for WPILIB to report infinite recursion?

auscompgeek, 2019-02-12:
Which language? Java would throw a StackOverflowError on infinite recursion, which should give you a traceback in the console.

C++ would probably segfault... I'm not sure how segfaults are reported to the console (or whether they even are).

rselwyn, 2019-02-12:
We are using Java, and there wasn't a traceback in the console.  It's part of the reason why it took us on the order of hours to fix.

PeterJohnson, 2019-02-12:
I saw this with a team last year, and I think it's a bug in netcomm where a too-large message doesn't make it to the netconsole stream.

AustinShalit, 2019-05-19:
Solved by #1618

closed 2019-05-19 by AustinShalit