wpilibsuite/allwpilib #1603: Deploy success but NO ROBOT CODE

WuYuanhun, 2019-02-16:
I successfully Deploy robot code to Roborio but the driver station says there is no robot code.

This has been occurred since 5 hours ago. And nothing is wrong before that.

AustinShalit, 2019-02-16:
Can you please post the output from the console in the driver station? You can open the console by pressing the gear in the top right and selecting open console.

SamCarlberg, 2019-02-16:
"No robot code" implies that your code is crashing. Do you see any errors in RIOLog or the driver station?

PeterJohnson, 2019-02-16:
What does the console output in the DS or riolog contain?  There should be a stack trace or messages there.

WuYuanhun, 2019-02-16:
> "No robot code" implies that your code is crashing. Do you see any errors in RIOLog or the driver station?

I used another computer to deploy the same code and there is nothing wrong.

AustinShalit, 2019-02-16:
If you try to use the computer that was not working again, paste the console output here.  Without the output we have very little information.

WuYuanhun, 2019-02-16:
> If you try to use the computer that was not working again, paste the console output here. Without the output we have very little information.

Thanks. I'm trying to get it cause right now I do not have a PC nearby.

WuYuanhun, 2019-02-16:
Problem Solved by fixing the code (change the order of subsystem definition).
I define Pneumatics system after Drive system which has call a member variable (solenoid) in Drive's init. The crash  should be occurred when drive system trying to call a non-set solenoid variable in Pneumatics.
Which makes sense that error message is never display in RioLog nor drive station, for it might be a runtime error.

Thanks for you guys!

closed 2019-02-16 by WuYuanhun