wpilibsuite/allwpilib #1606: Weird button press

NamelessSuperCoder, 2019-02-18:
Hello, I was trying to use the JoystickButton to be able to toggle something simple, and for some reason, it seemed to stutter, when I looked into it it was because the button was rebounding. This was something it never did before! I am using the 2019.3.2 version of the library. I know that I could implement some kind of method to account for this rebound but this should not be happening. We are using a Logitech and on the DriverStation on the USB port window, we can see that key works properly. Here is an implementation of we are trying to do.

In OI.java
`public static final JoystickButton button = new JoystickButton(gamepad, _3);`

In Robot.java

```
boolean canHatch = false;
public void teleopPeriodic() {
    Scheduler.getInstance().run();
    if (button.get()) {
      catchHatch = !catchHatch;
    }

    if (catchHatch) {
      if (hatchIntake.get()) {
        hatchIntake.set(false);
      }
    } else {
      if (!hatchIntake.get()) {
        hatchIntake.set(true);
      }
    }
  }
```

PeterJohnson, 2019-02-18:
Think about what happens when the button is pressed for more than one cycle through this loop.  You should store the previous state of the button and check for a rising edge only (e.g. the transition from not pressed to pressed) to swap the flag.

closed 2019-02-18 by NamelessSuperCoder

NamelessSuperCoder, 2019-02-18:
Thank you so just to make sure would this is a correct way of implementing the toggle then?
'button.whenPressed(new InstantCommand(() -> catchHatch = !catchHatch));'