wpilibsuite/allwpilib #1608: Shuffleboard add methods should be able to take supplier functions

SamCarlberg, 2019-02-18:
Instead of requiring users to keep track of NetworkTable entries even if they just want to have it update periodically, methods should be added so that users can provide a lambda or method reference to some function that gets the value they want to have in the widget.

### Current API
```java
// Have to keep track of the entry, AND specify a default value
NetworkTableEntry speedEntry = Shuffleboard.getTab("Tab")
  .add("Motor Speed", 0)
  .getEntry();

// Later, in a periodic update function
speedEntry.setNumber(myMotorController.get());
```

### Proposed API
```java
// Set and forget
Shuffleboard.getTab("Tab")
  .add("Motor Speed", myMotorController::get);
```

Java would have these 7 methods added to `ShuffleboardComponent` for the primitive types supported by NetworkTables

```
SimpleWidget add(String title, DoubleSupplier valueProvider)
SimpleWidget add(String title, BooleanSupplier valueProvider)
SimpleWidget add(String title, Supplier<String> valueProvider)
SimpleWidget add(String title, Supplier<double[]> valueProvider)
SimpleWidget add(String title, Supplier<boolean[]> valueProvider)
SimpleWidget add(String title, Supplier<String[]> valueProvider)
SimpleWidget add(String title, Supplier<byte[]> valueProvider)
```

Oblarg, 2019-02-27:
It'd be nice to have this supported directly by NetworkTableEntry, too, to be honest.

SamCarlberg, 2019-02-27:
It doesn't make much sense for NetworkTableEntry, since it'd need an update loop to continually poll the supplier.

Oblarg, 2019-02-27:
NetworkTable could just expose an "updateEntries" method.

PR #1690 (2019-05-30): Allow widgets to be added by passing value suppliers
Closes #1608 

This greatly reduces the ceremony required to simply push values to Shuffleboard.

```java
Gyro gyro;

// Old
NetworkTableEntry angleEntry = Shuffleboard.getTab("Tab")
  .add("Angle", 0)
  .getEntry();

@Override
public void periodic() {
  angleEntry.setDouble(gyro.getAngle());
}

// New
Shuffleboard.getTab("Tab")
  .addNumber("Angle", gyro::getAngle);
```
closed 2019-05-30 by PeterJohnson