wpilibsuite/allwpilib #1609: Shuffleboard layout properties bug

OmerWexler, 2019-02-18:
Hello, I have found a bug in the method

>  withProperties() 
in which the 

> builtInLayouts.kgrip 
would simply keep all the default values...
please attend, thanks ahead,
Omer Wexler, BumbleB #3339.

OmerWexler, 2019-02-21:
We have also found recently that the row by collumn loc on the grid of an opened tab (through robot code) is inconsistent, please try to attend to it soon.
Thank you.

auscompgeek, 2019-02-22:
Are you sure you have an up-to-date version of Shuffleboard?

OmerWexler, 2019-02-22:
We do constantly update our laptops to make sure we stay up to date so I would say so... I can check though, will reply soon.

OmerWexler, 2019-02-22:
Yes, I had the latest (2019.1.0) version installed.

ThadHouse, 2019-02-22:
Thats not the latest version. The latest version is 2019.3.1, which you can get from installing wpilib 2019.3.2.

OmerWexler, 2019-02-22:
Oh... my bad...

OmerWexler, 2019-02-22:
I will install the update on my PC and would check about all our team's laptops. 
Thank you.

closed 2020-09-13 by AustinShalit