wpilibsuite/allwpilib #1611: Request: Prefer protected over private for class members

fletch3555, 2019-02-19:
Using "private" scope prevents extending classes, causing teams to have to re-implement the class in it's entirety.

To give a more concrete example, I'm working with the MecanumDrive class, but need to configure it to use closed-loop control for the TalonSRX objects.  Since DriveCartesian calls `.Set(<value>)`, not `.Set(ControlMode::Velocity, <value>)`, I need to override that method.  When doing so, I copy/paste the method body into the subclass and recompile.  However, all members of the MecanumDrive class are private, most importantly the SpeedController objects.

I propose using `protected` scope instead.  This should apply to any class that is likely to be used and overridden by end users, including motor controllers, sensors, etc.

I was testing C++ specifically, but I imagine the same problem would occur in Java as well.

fletch3555, 2019-02-19:
Is there a reason this was closed without comment? I don't see anything in https://github.com/wpilibsuite/allwpilib/blob/master/CONTRIBUTING.md that my request needs to follow any particular format.  At the very least, a simple comment explaining what I did wrong would be appreciated.

calcmogul, 2019-02-19:
My phone was in my pocket when this PR was closed, so I didn't actually intend to close it. My bad.

calcmogul, 2019-02-19:
Composition would be preferred over inheritance for providing customization points. By making these classes take joystick inputs and return kinematic results, the user can pass them to whatever they want, not just motors (for example, it makes closed loop velocity control trivial). The alternative is the ClosedLoopMotor PR, but I dislike how it composes and there's a lot of conversion factors for normalization that don't need to exist.

It would be a breaking change done over the next two years, but makes more powerful abstractions possible in a much cleaner, linear fashion (swerve drive, closed loop MIMO control, etc.).

fletch3555, 2019-02-20:
That's a new one.  Never seen anyone "butt-close-github-issues" before. 😆

I like the idea of composition, but as you said, that's breaking and would require a couple years to phase it in.  Changing private to protected would not be breaking and could even be released this year (assuming another release was planned).

I'm willing to put together the PR for it, and will even cover both java and C++ sides, but would appreciate help building a list of classes this would be beneficial for.  I'm happy with leaving it to just the "drive" classes for now if that's reasonable?

Starlight220, 2020-03-25:
Just an idea : having the motor output to a lambda, so there is no need for overriding.

auscompgeek, 2020-03-27:
Given that the kinematics classes are in WPILib now, do those solve your usecase @fletch3555?

calcmogul, 2020-08-26:
The drive class use-case mentioned here would be resolved by the suggestions made in #1691.

closed 2021-03-13 by AustinShalit