wpilibsuite/allwpilib #1612: Missing command templates for Conditional Command

sciencewhiz, 2019-02-24:
There isn't a template for a ConditionalCommand in either wpilibjExamples or wpilibcExamples.

Starlight220, 2020-09-14:
What is wanted here? I see multiple options:
- More documenting of ConditionalCommand: if so, this issue should be moved to frc-docs
- An example using ConditionalCommand
- A VS Code file generator template
- something entirely different?

If this is resolved because of the command-based rewrite, close this.

AustinShalit, 2020-09-17:
> An example using ConditionalCommand

Do this one

sciencewhiz, 2020-10-01:
> A VS Code file generator template

This was the intent. VS Code pulls the templates from https://github.com/wpilibsuite/allwpilib/tree/master/wpilibjExamples/src/main/java/edu/wpi/first/wpilibj/commands

The FrisbeeBot example uses a ConditionalCommand, so that isn't needed.

Starlight220, 2020-10-01:
There are other command types that don't have a template: 

- ConditionalCommand
- CommandGroupBase
- FunctionalCommand
- MecanumControllerCommand
- NotifierCommand
- PIDCommand (New)
- PerpetualCommand
- PrintCommand
- ProxyScheduleCommand
- RamseteCommand
- RunCommand
- ScheduleCommand
- SelectCommand
- StartEndCommand
- SwerveControllerCommand
- WaitCommand
- WaitUntilCommand

However, not much use comes from overriding them...

rzblue, 2023-11-09:
This PR isn't irrelevant anymore- in the new command library ConditionalCommand isn't intended to be subclassed.

closed 2023-11-09 by rzblue