wpilibsuite/allwpilib #1619: Problems building allwpilib 2019.4.1

webbbt456, 2019-03-01:
I am trying to compile allwpilib 2019.4.1. I have compiled all of the previous version using VS2017 with out any problems.

In a command window, I type the following commands from within the allwpilib folder:

set JAVA_HOME=C:\Users\Public\frc2019\jdk
gradlew.bat -build

The following are the results:

FAILURE: Build failed with an exception.

* What went wrong:
The specified build file 'D:\FIRST\allwpilib\uild' does not exist.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 2s

I rename the allwpilib folder with out the version so I do not have to change my VS2017 x64 projects configuration every time an update is issued.  I rename the previous version folder with the version number and unzip the update and then rename it to allwpilib to avoid mixing files from previous version.

ThadHouse, 2019-03-01:
You just want to be using `gradlew.bat build`, not `gradlew.bat -build`. The second one sets the input folder to a wrong folder, which is the error you're seeing.

webbbt456, 2019-03-01:
Thanks, that was the problem.

closed 2019-03-01 by webbbt456