wpilibsuite/allwpilib #1620: Don't check enabled separately

virtuald, 2019-03-01:
The current dominant pattern tends to be `isEnabled() && isOperatorControl()` or some equivalent, but each of these takes a lock and checks a cached value. It would be more efficient to have functions that do what you want to avoid the extra lock contention.

* isOperatorControlEnabled
* isAutonomousEnabled

Another thing could be to change the existing isOperatorControl/isAutonomous functions to also check enabled.

AustinShalit, 2019-03-01:
I prefer option 2. I think it is confusing that a robot can be in autonomous but also be disabled. But doing option 2 would be breaking. Maybe we can overload with a boolean to check if enabled?

ThadHouse, 2019-03-01:
They're not mutually exclusive. Enabled disabled is a separate bit then auto teleop, and there is a difference between auto disabled and teleop disabled.

virtuald, 2019-03-01:
I'm going to add isOperatorControlEnabled/isAutonomousEnabled to robotpy this season because profiling has shown its a source of overhead/lock contention for me.

calcmogul, 2019-03-01:
TimedRobot treats them as mutually exclusive though, and so do users by extension.

ThadHouse, 2019-03-01:
We used the flag inside of disabled 1 year to check though.

PR #2661 (2020-08-29): [wpilib] Add methods to check game and enabled state together
Closes #1620

This avoids users having to call `IsOperatorControl() && IsEnabled()` to figure out if their robot is enabled and in the teleop state. The expression above involves calling two methods that each have their own lock.

These new methods should only involve locking one mutex, since only one call is made to `HAL_GetControlWord()`.
closed 2020-08-29 by PeterJohnson