wpilibsuite/allwpilib #1622: Document that solenoid calls could take a potentially unbounded period of time

virtuald, 2019-03-02:
Per my conversation with @ThadHouse last night, he said that perhaps calls to solenoid get/set head out to the CAN bus, and could potentially take a unexpectedly long period of time to complete.

This should be added to the solenoid documentation if this is indeed the case. Ideally it should also be documented how long it would be expected to take.

virtuald, 2019-03-10:
There does appear to be a timeout of some kind, so it's probably not unbounded. But that timeout should probably be documented too.

kasemir, 2019-03-14:
https://www.chiefdelphi.com/t/we-are-experiencing-a-lag-issue-in-our-code/350295/6 : 
We have some solenoids in our robot which are only occasionally actuated, so figured they won't use any resources.
Then found that simply having a solenoid means that the LiveWindow is querying them every update period, and that ended up consuming 70% of the loopFunc().
 LiveWindow.disableAllTelemetry() fixed that, but again it was unexpected.

Starlight220, 2020-04-20:
> Then found that simply having a solenoid means that the LiveWindow is querying them every update period, and that ended up consuming 70% of the loopFunc().
> LiveWindow.disableAllTelemetry() fixed that, but again it was unexpected.

That should be added to frc-docs, and perhaps to some other documentation.

ThadHouse, 2020-04-20:
This is actually not applicable as of the 2020 season. With the CAN rewrite, these calls are deterministic now.

closed 2020-04-20 by ThadHouse