wpilibsuite/allwpilib #1623: Watchdog is too complicated

virtuald, 2019-03-02:
Blah blah blah backwards compat but... 

I really really like the addition of the watchdog this year. It's really useful, and provides a needed warning to teams when their loops are getting out of hand. So far, so good. But it does too much (and at least in RobotPy it has a lot of unnecessary overhead).

In particular, I think it should be changed to be much more lightweight (or provide a lightweight version) and only support operation in a single thread -- no locks, and do the expiry computation inline. As far as I can see, the only value that operating a central scheduler thread provides is detecting infinite loops -- any other case will work just fine. There's probably a simpler solution for detecting infinite loops too.

calcmogul, 2019-03-04:
Detecting and warning about infinite loops requires a separate thread that complains on timeout. I don't see a way around that, so the thread and synchronization are necessary overhead. Motor safety does precisely this via the DriverStation thread. The intent was to replace the DS thread thing with Watchdog, but CTRE reported bugs so the PR that added it got partially reverted. I haven't had time to ressurect it into a cohesive PR and investigate the issue.

However, a separate synchronous tracing class could be useful in general, if its users are well behaved. It would be a wrapper around a Timer that keeps a vector of epochs. The API would be the same as Watchdog. Idk whether to call it Tracer or Profiler. I guess the latter depends on what else it can profile. In the future, I could see this being used in Notifier to warn users of overruns.

virtuald, 2019-03-04:
One way to get around the mutex synchronization is to use an atomic incrementing counter -- since if the point is to detect an infinite loop (as non-infinite loops would already be detected inline), it could be checked significantly less often (say every 5 seconds).

Starlight220, 2020-04-20:
> However, a separate synchronous tracing class could be useful in general, if its users are well behaved. It would be a wrapper around a Timer that keeps a vector of epochs. The API would be the same as Watchdog. Idk whether to call it Tracer or Profiler; I guess the latter depends on what else it can profile. In the future, I could see this being used in Notifier to warn users of overruns or provide timing data in general.

This was fixed with #2452. Is that enough to close this issue, or are there any plans for this?

calcmogul, 2020-08-26:
Resolved by a combination of #2452, #2456, and #2602.

closed 2020-08-26 by calcmogul