wpilibsuite/allwpilib #1629: Windows installer download problems

jmllinger, 2019-03-09:
The release files are huge and currently need to be downloaded in one fell swoop.  It would be nice to have a download helper like NI uses that could automatically resume a download that gets interrupted.  At present, my browser estimates about 2 1/2 hrs for the download, and usually fails before finishing.  It has taken me several days of retrying the download before I can get it successfully. (Windows 7 64-bit, Firefox (latest)).  This is especially problematic for "required" updates in the middle of build and competition season!

An alternative would be to provide several smaller installers for those who can't get the huge one.

PeterJohnson, 2019-03-09:
Have you tried using a download manager such as uGet?  https://ugetdm.com/

jmllinger, 2019-03-09:
No, I didn't know there were such things.  I'll give it a try.

AustinShalit, 2019-05-19:
Please reopen this issue if you are still having trouble downloading the installer.

closed 2019-05-19 by AustinShalit