wpilibsuite/allwpilib #1631: set Scheduler.m_disabled initial value to true

worthp, 2019-03-10:
It seems that my TimedRobot subclass is responsible for maintaining the Scheduler enable/disable state. Where I may slightly disagree with having my bot code responsible for maintaining Scheduler state when RobotBase classes are a better way to synchronize state from a safety perspective, I would strongly suggest that the default value of **Scheduler.m_disabled** should be true. 

We have discovered that pushing a button, with a Command bound on WhenPressed while the bot was disabled, was running the command (per Command based example) that controlled out elevator to go up to a position. That command was being called but not causing the motors to move because the bot was disabled. As soon as a match was started the bot was enabled, our elevator shot to the top during autonomous. 

If teams are responsible for enabling/disabling Scheduler, it should not be enabled by default. It is a very dangerous default behavior when teams are not aware they are responsible. Separately, I'd suggest that Scheduler enabled state should be maintained in the RobotBase classes. Even more separately, the Command based example Robot code should explicitly show enable/disable calls to Scheduler in the xyzInit methods and an explicit call to disable() in disablePeriodic() methods if this design remains.

Kevin-OConnor, 2019-03-10:
The scheduler is intended to run when the robot is disabled. That is why there is a Scheduler run call in the disabledPeriodic methods of the command templates. Many teams want their commands to still run while disabled. If you don't, remove that call or disable the scheduler yourself.

There is almost never a reason to disable the scheduler. It is only needed if you want to dynamically disable it from somewhere other than where the Scheduler Run call is being made.

worthp, 2019-03-10:
I understand the design point. I am not debating that in this issue. I disagree, but not debating here. What i am calling out here is that here is that there  some dangerous side effects as I explained when a command is running and the isFinished() is waiting for hardware to move that will complete only when enabling. This design and control mechanism needs to be documented and made clear. One way to make it clear is to force my Robot code to explicitly enable the Scheduler (defaulting to m_disabled = true will do that). I can do that in robotInit() and easy to understand. But having commands being able to run in the background controlling hardware while the bot is disabled is dangerous and needs to be well understood and documented.

By accident, someone pressed a button whose command started running and continued to run while disabled but had no effect because the hardware would not respond because the bot was disabled. As soon as bot was enabled the command ran to completion. This behavior and some unintended consequences need to be documented somewhere in very clear terms. At least they need to be called in in example templates.

worthp, 2019-03-10:
Furthermore, 
https://github.com/wpilibsuite/allwpilib/blob/master/wpilibj/src/main/java/edu/wpi/first/wpilibj/command/Command.java#L279
/**
   * Sets whether or not this {@link Command} should run when the robot is disabled.
   *
   * <p>**By default a command will not run when the robot is disabled**, and will in fact be canceled.
   *
   * @param run whether or not this command should run when the robot is disabled
   */
  public void setRunWhenDisabled(boolean run) {
    m_runWhenDisabled = run;
  }

This is inconsistent documentation and a very dangerous one at that. Please at least change the default value. This is asking for a serious injury or damage.

sciencewhiz, 2019-03-10:
The run method of the command checks if the robot is disabled and if m_runWhenDisabled is false it cancels the command, which is exactly what the documentation says. https://github.com/wpilibsuite/allwpilib/blob/655763a9a7d55b1618d4df6f1b2f606ff16c04dc/wpilibj/src/main/java/edu/wpi/first/wpilibj/command/Command.java#L279

It does this because the scheduler is running, not in spite of it.

worthp, 2019-03-10:
To be clear, i am not fighting with you guys. i know the code base is huge and there is a lot to understand. Once we saw the behavior and posited the theory that a drive team member accidentally pressed the elevator control button while the bot was disabled (sitting in Q waiting for the match) and upon enabling, the elevator went to the desired position. btw, we use TalonSRX motion magic.

what i can tell you for sure is that after we saw the symptom we went the pit immediately after the match and:
- powered down the bot
- powered up the bot
- we waited for comms
- before enabling, we pressed the button that tells our talon srx controllers to use Motion Magic to go to a position 
- upon enabling the elevator went to that position

we then put this line in the execute() of the Command:
   if (DriverStation.getInstance().isDisabled()) return;

The behavior stopped. We will look into it more, but i am sure there is something weird going on. When i was tracing through i got into the Scheduler.enabled thing which is how i got to this issue. After the back and forth on this issue, perhaps the confusion is somewhere else.

Kevin-OConnor, 2019-03-11:
If you are operating in way that robot movement while enabled can cause "Serious injury or damage" you should change your operating procedures. While the robot is enabled you should act like any part of the robot could move at any speed at any time. The Disable/E-stop is the tested safety functionality of the software.

worthp, 2019-03-11:
I will find the bug in the Command, fix it and create a proper PR

closed 2019-03-11 by worthp

virtuald, 2019-03-11:
While I think @worthp's solution to the bug is incorrect, I smell a bug here.

In `run`, Line 280 will call cancel -- but cancel will only cancel a command if the command is running. In line 282 that checks isCanceled, will initially return false, because it was never canceled. 

If I'm reading this correctly (I've never used commands before), that means when the robot first starts up each command could potentially be initialized and executed at least once in disabled mode if something asks the command to run (like a button)? That would be contrary to the documentation and explain the behavior that @worthp is seeing.

This should be easy enough to check (not it!).

worthp, 2019-03-11:
@virtuald Thank you.

worthp, 2019-03-11:
What I will be looking into tonight is the difference between IterativeRobot:
  public void startCompetition() {
    robotInit();

    // Tell the DS that the robot is ready to be enabled
    HAL.observeUserProgramStarting();

    // Loop forever, calling the appropriate mode-dependent function
    while (true) {
      // Wait for new data to arrive
      m_ds.waitForData();

      loopFunc();
    }
  }

and TimedRobot:

@SuppressWarnings("UnsafeFinalization")
  public void startCompetition() {
    robotInit();

    // Tell the DS that the robot is ready to be enabled
    HAL.observeUserProgramStarting();

    m_expirationTime = RobotController.getFPGATime() * 1e-6 + m_period;
    updateAlarm();

    // Loop forever, calling the appropriate mode-dependent function
    while (true) {
      long curTime = NotifierJNI.waitForNotifierAlarm(m_notifier);
      if (curTime == 0) {
        break;
      }

      m_expirationTime += m_period;
      updateAlarm();

      loopFunc();
    }
  }